carla-simulator/carla #3543: manual_control.py uses hard-coded port 8000 for TM

Kipt, 2020-11-04:
I'm running several Carla instances through Docker and map the Traffic Manager port, but _manual_control.py_ includes the _manual_control.py_ in the examples folder that uses default port. This causes a runtime error in `KeyboardControl` constructor, `world.player.set_autopilot(self._autopilot_enabled)`.

Steps to reproduce:
1. Run Carla in Docker and don't expose port 8000.
2. Run scenario runner with appropriate `--host`, `--port` and `--trafficManagerPort` arguments.
3. Run scenario runner's `manual_control.py`.

**Expected behavior**
pygame window appearing with a vehicle to control.

**Actual behavior**
Crash with message:
```
Traceback (most recent call last):
  File "manual_control.py", line 220, in main
    game_loop(args)
  File "manual_control.py", line 148, in game_loop
    controller = KeyboardControl(world, args.autopilot)
  File "/home/adam/carla_apis/PythonAPI_v0.9.10/examples/manual_control.py", line 292, in __init__
    world.player.set_autopilot(self._autopilot_enabled)
RuntimeError: trying to create rpc server for traffic manager; but the system failed to create because of bind error.
```

dhilazo, 2020-11-05:
A similar error is found when executing the tutorial.py from the PythonAPI examples. In my case I runned the client on localhost and port 2000.
```
~]$ python tutorial.py
created vehicle.audi.etron
destroying actors
Traceback (most recent call last):
  File "tutorial.py", line 76, in main
    vehicle.set_autopilot(True)
RuntimeError: trying to create rpc server for traffic manager; but the system failed to create because of bind error.
```
Hope it helps!

qhaas, 2020-12-03:
Seeing this in CARLA 0.9.8 when I run more than one CARLA PythonAPI client script

EsquilliDev, 2021-01-19:
@qhaas I'm experiencing the same issue. Any fixes? So far I have to restart PC, which is a hassle.

qhaas, 2021-02-03:
No, one is going to have to edit the C++ side of the TM code to address this, I believe.

EsquilliDev, 2021-02-03:
My fix so far is to kill all Python process' and run the code again.

corkyw10, 2021-02-25:
@jackbart94 can you have a look at this please?

jackbart94, 2021-03-10:
Please, could you let me know if this is happening with current dev, or at least CARLA 0.9.11?

qhaas, 2021-03-10:
Here is a simple test with CARLA 0.9.11, seems to keep on using port 8000:
```
#!/bin/python3

import socket
import traceback
import time

import carla

try:
    client = carla.Client('127.0.0.1', 2000)
    tm = client.get_trafficmanager(6000)
    world = client.get_world()
    blueprint_library = world.get_blueprint_library()
    transform = world.get_map().get_spawn_points()[0]
    vehicle = world.spawn_actor(blueprint_library.filter('vehicle')[0], transform)
    vehicle.set_autopilot(True)

    time.sleep(1)
    for i in [22,3000,6000,8000]:
        testSocket = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
        result='inactive'
        if testSocket.connect_ex(('127.0.0.1',i)) == 0:
            result='active'
        print('{}: {}'.format(i,result))
        testSocket.close()
except:
    traceback.print_exc()
```
The results with CARLA 0.9.11:
```
22: active
3000: inactive
6000: active
8000: active
```

closed 2021-11-18 by jackbart94