carla-simulator/carla #3544: AttributeError: 'ServerSideSensor' object has no attribute 'bounding_box'

FHof, 2020-11-04:
### Problem Description

When I use the latest Carla Rosbridge with Carla 0.9.10 or 0.9.10.1, the Rosbridge crashes and shows a missing bounding box error:
```
  File "/home/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/traffic_participant.py", line 101, in get_object_info
    self.carla_actor.bounding_box.extent.x * 2.0,
AttributeError: 'ServerSideSensor' object has no attribute 'bounding_box'
```
Here is the full output: [bounding_box_error.zip](https://github.com/carla-simulator/carla/files/5487591/bounding_box_error.zip)
`ServerSideSensor` is a C++ class in LibCarla ([here](http://carla.org/Doxygen/html/de/d68/classcarla_1_1client_1_1ServerSideSensor.html) is its reference) and does not appear in the Rosbridge code, i.e. searching for `ServerSideSensor` in the repository files gives no result. Therefore I assume that the problem is related to the Carla server and/or client API.

### How to Reproduce

* Start the official 0.9.10 Carla server docker container
* Start a client docker container with ROS (e.g. noetic), the Carla API and latest Carla Rosbridge installed
* Execute following command (in the container with the Rosbridge): `roslaunch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch`

The problem does not occur with Carla 0.9.9 (and PPA Rosbridge version).
Furthermore, for some reason the problem did not happen when someone else tried to reproduce it on his own host with his Carla server.

### Additional Information

The problem happens with several versions of Carla, the Rosbridge and ROS
**Carla versions:** 0.9.10 (docker image), 0.9.10.1 (docker image), 0.9.10.1 (from releases, running on host)
**Rosbridge versions:** ee524bb50a, cb83740c5af49f
**ROS versions:** melodic, noetic
**Platforms:** A Docker image based on ros:noetic-robot for the clients, the official Carla docker images, and the host OS `Ubuntu 20.04.1 LTS (GNU/Linux 5.4.0-45-generic x86_64)`

corkyw10, 2021-03-01:
@joel-mb could you follow up on this please?

FHof, 2021-03-01:
The problem no longer happens with CARLA 0.9.11.

closed 2021-03-01 by FHof