carla-simulator/carla #3547: CARLA "MAKE LAUNCH" ISSUE

Yerin703, 2020-11-05:
make launch
Setup.sh: llvm-8.0 already installed.
Setup.sh: Retrieving boost.
--2020-11-05 18:16:43--  http://dl.bintray.com/boostorg/release/1.72.0/source/boost_1_72_0.tar.gz
Resolving dl.bintray.com (dl.bintray.com)... 54.187.47.108, 52.43.200.1, 52.88.32.158, ...
Connecting to dl.bintray.com (dl.bintray.com)|54.187.47.108|:80... connected.
HTTP request sent, awaiting response... 302 
Location: http://d29vzk4ow07wi7.cloudfront.net/c66e88d5786f2ca4dbebb14e06b566fb642a1a6947ad8cc9091f9f445134143f?response-content-disposition=attachment%3Bfilename%3D%22boost_1_72_0.tar.gz%22&Policy=eyJTdGF0ZW1lbnQiOiBbeyJSZXNvdXJjZSI6Imh0dHAqOi8vZDI5dnprNG93MDd3aTcuY2xvdWRmcm9udC5uZXQvYzY2ZTg4ZDU3ODZmMmNhNGRiZWJiMTRlMDZiNTY2ZmI2NDJhMWE2OTQ3YWQ4Y2M5MDkxZjlmNDQ1MTM0MTQzZj9yZXNwb25zZS1jb250ZW50LWRpc3Bvc2l0aW9uPWF0dGFjaG1lbnQlM0JmaWxlbmFtZSUzRCUyMmJvb3N0XzFfNzJfMC50YXIuZ3olMjIiLCJDb25kaXRpb24iOnsiRGF0ZUxlc3NUaGFuIjp7IkFXUzpFcG9jaFRpbWUiOjE2MDQ1Njg1MjN9LCJJcEFkZHJlc3MiOnsiQVdTOlNvdXJjZUlwIjoiMC4wLjAuMC8wIn19fV19&Signature=aTN4WXUoe1jsbttygQay6hClnOmDXAeNCdvAZeGn2qT40QrzciXyphCOXufAG-eXyDynSaPTrg3-Aid3n9wbejgldS30W5nDLbpILIFuhgK7-xkUOe~NVMBCoxLoSLpBmNUegR3c2m4Ap4JBe2Mz2DjFc0Xism7tOHGvTFBh2g2hnk2HOGC8cqVVA-dATBlb2X~qB9IiMvL~7HdaQSwRoD27n3xg9Ggd3Lfu3-kmgkTpd1aYh5d6AUHMlK4Sp4Yn7KpTEt6kK3fWlVdw1trA7LyYYmxyUDDo6ink4jK4SBQ0gHLhVx3MEY-LxMT~Zub0sc4f5swg6FPyepO~oj-EWg__&Key-Pair-Id=APKAIFKFWOMXM2UMTSFA [following]
--2020-11-05 18:16:43--  http://d29vzk4ow07wi7.cloudfront.net/c66e88d5786f2ca4dbebb14e06b566fb642a1a6947ad8cc9091f9f445134143f?response-content-disposition=attachment%3Bfilename%3D%22boost_1_72_0.tar.gz%22&Policy=eyJTdGF0ZW1lbnQiOiBbeyJSZXNvdXJjZSI6Imh0dHAqOi8vZDI5dnprNG93MDd3aTcuY2xvdWRmcm9udC5uZXQvYzY2ZTg4ZDU3ODZmMmNhNGRiZWJiMTRlMDZiNTY2ZmI2NDJhMWE2OTQ3YWQ4Y2M5MDkxZjlmNDQ1MTM0MTQzZj9yZXNwb25zZS1jb250ZW50LWRpc3Bvc2l0aW9uPWF0dGFjaG1lbnQlM0JmaWxlbmFtZSUzRCUyMmJvb3N0XzFfNzJfMC50YXIuZ3olMjIiLCJDb25kaXRpb24iOnsiRGF0ZUxlc3NUaGFuIjp7IkFXUzpFcG9jaFRpbWUiOjE2MDQ1Njg1MjN9LCJJcEFkZHJlc3MiOnsiQVdTOlNvdXJjZUlwIjoiMC4wLjAuMC8wIn19fV19&Signature=aTN4WXUoe1jsbttygQay6hClnOmDXAeNCdvAZeGn2qT40QrzciXyphCOXufAG-eXyDynSaPTrg3-Aid3n9wbejgldS30W5nDLbpILIFuhgK7-xkUOe~NVMBCoxLoSLpBmNUegR3c2m4Ap4JBe2Mz2DjFc0Xism7tOHGvTFBh2g2hnk2HOGC8cqVVA-dATBlb2X~qB9IiMvL~7HdaQSwRoD27n3xg9Ggd3Lfu3-kmgkTpd1aYh5d6AUHMlK4Sp4Yn7KpTEt6kK3fWlVdw1trA7LyYYmxyUDDo6ink4jK4SBQ0gHLhVx3MEY-LxMT~Zub0sc4f5swg6FPyepO~oj-EWg__&Key-Pair-Id=APKAIFKFWOMXM2UMTSFA
Resolving d29vzk4ow07wi7.cloudfront.net (d29vzk4ow07wi7.cloudfront.net)... 13.224.8.153, 13.224.8.28, 13.224.8.33, ...
Connecting to d29vzk4ow07wi7.cloudfront.net (d29vzk4ow07wi7.cloudfront.net)|13.224.8.153|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 126580835 (121M) [application/gzip]
Saving to: ‘boost_1_72_0.tar.gz’

boost_1_72_0.tar.gz 100%[===================>] 120.72M  4.71MB/s    in 28s     

2020-11-05 18:17:12 (4.31 MB/s) - ‘boost_1_72_0.tar.gz’ saved [126580835/126580835]

Setup.sh: Extracting boost for Python 3.
Building Boost.Build engine with toolset clang... 
Failed to build Boost.Build build engine
Consult 'bootstrap.log' for more details
Util/BuildTools/Linux.mk:137: recipe for target 'setup' failed
make: *** [setup] Error 1

We have problem in "make launch".
I have no idea how to fix it.

PLEASE HELP ME.

Yerin703, 2020-11-05:
***********
Ubuntu 18.04
clang version 8
gcc 7.5.0
python 3.7
************

Double-Xiong, 2021-01-11:
the same problem

Double-Xiong, 2021-01-18:
> the same problem 

I removed my anaconda3，it works。maybe the Python include path error

corkyw10, 2021-03-01:
@Yerin703 are you still experiencing this issue?

Yerin703, 2021-03-01:
No, I'm OK with that problem.

closed 2021-03-01 by corkyw10

SarathHari90, 2021-08-21:
> […](#)

Can you tell how did you resolve this issue?

dengbin1010, 2021-09-07:
> > […](#)
> 
> Can you tell how did you resolve this issue?

Hi bro, can u tell me how did u solve this issue? I meet the same problem! thank u

jjeong19, 2021-09-09:
The bootstrap.log can be found in Build/boost-1.72.0-c8-source/ 
i believe. After checking that, it showed an error regarding the string declaration. 
#https://github.com/boostorg/build/issues/569 and #https://github.com/boostorg/build/issues/468 talks about how the compiler uses system "strings.h" instead of the "strings.h" inside the jam. They resolve it by renaming the strings.h to jam_strings.h

Renaming and building it worked for me. Also if you straight up use the `make PythonAPI`, I believe it will  erase the source files in Build/boost-1.72.0-c8-source/. So what I did was I downloaded the source files separately and built it, and installed it in Build/boost-1.72.0-c8-install/ . Make sure to build boost 1.72 with the correct python version so that Setup.sh recognizes that it's there.

Hope this works for you!

BeiZhangChen, 2021-09-28:
Hey bro, what is the location of the string.h could not find it, thanks

martino92, 2022-03-31:
Is any solution about this problem? How can I reopen this issue?

dengbin1010, 2022-03-31:
已收到您的邮件！谢谢！

BepfCp, 2022-07-05:
I met this problem too. And below is my solution, which works for me.

1. Open Util/BuildTools/Setup.sh, go to line 154, modify the `py3` variable to your own python path, e.g.,
```bash
py3="/home/ubuntu/miniconda3/envs/carla/bin/python${PY_VERSION}"
```
2. Open ~/.bashrc, add CPLUS_INCLUDE_PATH, e.g,
```
export CPLUS_INCLUDE_PATH=~/miniconda3/envs/carla/include/python3.7m/
```
Make sure to change to your own pyconfig.h, you use `locate pyconfig.h` to identify it.

dengbin1010, 2022-07-05:
已收到您的邮件！谢谢！

Dragonking3x, 2022-10-10:
try to use
`sudo make launch`

it is strange, because you get an error with "sudo make launch" that says "UE4_ROOT variable is missing".

If start with
`make launch`
wait for the error and then write
`sudo make launch`
and after the next error finish with
`make launch`
it works

dengbin1010, 2022-10-11:
已收到您的邮件！谢谢！

jerry73204, 2022-11-12:
As @jjeong19 pointed out, I cannot build CARLA in dev branch for the same reason. My system is Ubuntu 20.04 using system default Pyhton 3.8.10. I packed the patched boost source code and those interested can download the file here.

https://github.com/jerry73204/carla/releases/tag/fix-boost

dengbin1010, 2022-11-12:
已收到您的邮件！谢谢！

jerry73204, 2022-11-12:
@bernatx Is it possible to host the patched boost on a CARLA website or ship the package in future releases?

zzj403, 2023-08-21:
> The bootstrap.log can be found in Build/boost-1.72.0-c8-source/ i believe. After checking that, it showed an error regarding the string declaration. #[boostorg/build#569](https://github.com/boostorg/build/issues/569) and #[boostorg/build#468](https://github.com/boostorg/build/issues/468) talks about how the compiler uses system "strings.h" instead of the "strings.h" inside the jam. They resolve it by renaming the strings.h to jam_strings.h
> 
> Renaming and building it worked for me. Also if you straight up use the `make PythonAPI`, I believe it will erase the source files in Build/boost-1.72.0-c8-source/. So what I did was I downloaded the source files separately and built it, and installed it in Build/boost-1.72.0-c8-install/ . Make sure to build boost 1.72 with the correct python version so that Setup.sh recognizes that it's there.
> 
> Hope this works for you!

very useful for me on ubuntu18+carla-.9.10.1+ue4.24, Thank you very much!

dengbin1010, 2023-08-21:
已收到您的邮件！谢谢！