carla-simulator/carla #3552: On map import: The following assets failed to save correctly: /Game/Carla/Maps/BaseMap/BaseMap

philippwulff, 2020-11-05:
I am trying to import a map into CARLA using "make import" and this is the import crashes, when it reaches this line:

`The following assets failed to save correctly: /Game/Carla/Maps/BaseMap/BaseMap`

This line is from the bottom of this [error log.txt](https://github.com/carla-simulator/carla/files/5495006/error.log.txt).

How do I fix this? The repo does not contain the BaseMap file, so that I could simply download that and exchange it for the one in this folder. Also I would uninstall and reinstall Carla and UE4, but I could not find instructions on how to do that in Linux safely.

corkyw10, 2021-03-02:
@philippwulff Thanks for your patience, are you still experiencing this issue?

@bernatx would you be able to follow up please?

AthulGangadharan, 2021-04-18:
Even I'm facing this issue, any leads? Please help

bernatx, 2021-04-19:
Hi,
From this line 

`[2020.11.05-13.36.39:544][  0]LogAssetTools: Warning: You can not create an asset named 'HRM_from_sumo_w_dummy' because there is already a map file with this name in this folder.
`

I see you are trying to import a map that already exists. That is not possible, you need to remove the map that already is imported before reimport it.

From this line

`1001[2020.11.05-14.20.47:797][  0]LogLinker: While loading '../../../../carla/Unreal/CarlaUE4/Content/Carla/Static/TrafficLight/Streetlights_01/TrafficLightPieceStructure.uasset' failed to load '/Game/Carla/Static/TrafficSigns/Streetlights_01/SM_TLights_Leds': Can't find file.
`

it seems that you don't have the content assets downloaded. To import a map, you need a working version of CARLA, that means you need to compile CARLA and UE4, and also download all the content assets.

Saquib764, 2021-04-28:
I am also getting the same error. 

I am following these instruction - https://carla.readthedocs.io/en/0.9.10/tuto_A_add_map/#map-ingestion-in-a-build-from-source

Using the following command in the above documentation doesn't work -
```
make import ARGS="--package package_name --no-carla-materials"
```
therefore I used - `make import` instead, which leads me to the error -
```
The following assets failed to save correctly: /Game/Carla/Maps/BaseMap/BaseMap
```

closed 2023-06-02 by stale[bot]