carla-simulator/carla #3555: Docker run Carla 0.9.10 crashes with Segmentation Fault error  VK_ERROR_INITIALIZATION_FAILED

XieJinLiang, 2020-11-06:
I use the Nvidia 2080 with 8G gpu memory.
And the Nvidia driver for me 450.66, OS is 16.04LTS.
I have docker pull carlasim/carla:0.9.10, 0.9.9, 0.9.6.

If **not** running carla0.9.10 in docker, carla server all going well

=====================================================================
When I run carla0.9.10 in docker
`nvidia-docker run -it --rm=true --name carla_server --net=host --runtime=nvidia -e NVIDIA_VISIBLE_DEVICES=0 -e DISPLAY=${DISPLAY} carlasim/carla:0.9.10 /bin/bash 
`
where $DISPLAY=:0
I enter the pseudo-input terminal,then I run `./CarlaUE4.sh`, the carla server window will flash by , then disappear and reported the following error

`4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
sh: 1: xdg-user-dir: not found
LowLevelFatalError [File:Unknown] [Line: 772] 
VulkanRHI::vkGetPhysicalDeviceSurfaceFormatsKHR(Device.GetPhysicalHandle(), Surface, &NumFormats, nullptr) failed, VkResult=-3
 at /home/jenkins/UnrealEngine_4.24/Engine/Source/Runtime/VulkanRHI/Private/VulkanSwapChain.cpp:152 
 with error VK_ERROR_INITIALIZATION_FAILED
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=93152 LargeMemoryPoolOffset=224288 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault
`

========================================================================
When I run carla0.9.9 in docker with similar command, reported:
`4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
sh: 1: xdg-user-dir: not found
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=94288 LargeMemoryPoolOffset=225424 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=81792 LargeMemoryPoolOffset=212928 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault
`

==========================================================================
But when I run carla0.9.6 in docker with similar command, only reported:
`4.22.1-0+++UE4+Release-4.22 517 0
Disabling core dumps.
sh: 1: xdg-user-dir: not found`

There is also xdg-user-dir errors, but carla server can starts normally, and the manula_control.py script can also run normally.

==========================================================================

rowandempster, 2020-11-07:
Docker carla server also crashes for me on 0.9.10 (works on 0.9.9), following.

rowandempster, 2020-11-07:
Found a working solution for me:

0.9.9 Dockerfile:
```
# Make sure drivers are >= 390
# docker run -p 2000-2002:2000-2002 --runtime=nvidia -e NVIDIA_VISIBLE_DEVICES=ID carla:latest ./CarlaUE4.sh
# -carla-rpc-port=2000 -carla-streaming-port=2001

FROM nvidia/vulkan:1.1.121-cuda-10.1-alpha

RUN packages='libsdl2-2.0' \
	&& apt-get update && apt-get install -y $packages --no-install-recommends \
	&& rm -rf /var/lib/apt/lists/*

RUN useradd -m carla

COPY --chown=carla:carla . /home/carla

USER carla
WORKDIR /home/carla

ENV SDL_VIDEODRIVER=offscreen

CMD /bin/bash CarlaUE4.sh```

0.9.10 Dockerfile: 
```# Make sure drivers are >= 390
#
# If you want the highest performance you need to make sure that your host has a functioning Xorg server running
# even in headless/cloud systems.
# If your host has a running version of Xorg then set  SDL_VIDEODRIVER=x11 and run the docker container as:
#
# sudo -E docker run --privileged --rm --gpus all -it --net=host -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix:rw carla:latest /bin/bash
#
# Otherwise, set SDL_VIDEODRIVER=offscreen and run the docker container as:
#
# sudo -E docker run --rm --gpus all -it --net=host carla:latest /bin/bash

FROM nvidia/vulkan:1.1.121-cuda-10.1--ubuntu18.04

RUN packages='libsdl2-2.0 xserver-xorg libvulkan1' \
	&& apt-get update && DEBIAN_FRONTEND=noninteractive apt-get install -y $packages --no-install-recommends \
    && VULKAN_API_VERSION=`dpkg -s libvulkan1 | grep -oP 'Version: [0-9|\.]+' | grep -oP '[0-9|\.]+'` && \
	mkdir -p /etc/vulkan/icd.d/ && \
	echo \
	"{\
		\"file_format_version\" : \"1.0.0\",\
		\"ICD\": {\
			\"library_path\": \"libGLX_nvidia.so.0\",\
			\"api_version\" : \"${VULKAN_API_VERSION}\"\
		}\
	}" > /etc/vulkan/icd.d/nvidia_icd.json \
	&& rm -rf /var/lib/apt/lists/*

RUN useradd -m carla

COPY --chown=carla:carla . /home/carla

USER carla
WORKDIR /home/carla

# Select the video driver between offsreen and x11
# x11 is recommended for vulkan support
ENV SDL_VIDEODRIVER=x11

# you can also run CARLA in -opengl mode
CMD /bin/bash CarlaUE4.sh -vulkan
```

Notice the change to the **SDL_VIDEODRIVER** environment variable. Changing it back to `offscreen` using 
```
 docker run
           -it \
	   -e SDL_VIDEODRIVER='offscreen' \
           carlasim/carla:0.9.10 \
           ./CarlaUE4.sh Town02 -quality-level=Low -carla-server
```
worked for me.

XieJinLiang, 2020-11-09:
@rowandempster Thank you!!
When I add this variable `-e SDL_VIDEODRIVER='offscreen',` there is indeed no more `VK_ERROR_INITIALIZATION_FAILED` error, but it will still report the same error as docker running 0.9.9
```
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
sh: 1: xdg-user-dir: not found
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=118144 LargeMemoryPoolOffset=249280 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=105648 LargeMemoryPoolOffset=236784 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault
carla@allride-ThinkStation-P720:~$ exit
exit
```

XieJinLiang, 2020-11-09:
@rowandempster 
Maybe I know what's going on. According to your idea, I compared their docker files and found that 0.9.9 and 0.9.10 both use nvidia/vulkan by default, while 0.9.6 uses nvidia/opengl. I added the parameter -opengl when starting carla,  it works!

closed 2021-03-03 by corkyw10