carla-simulator/carla #3557: When pressing play in UE4 Editor: Exception thrown: bind: Address already in use

philippwulff, 2020-11-09:
I reinstalled Carla and UE4. Now I get this error, when I press play in a map:

`Fatal error: [File:/home/marcel/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.cpp] [Line: 102] Exception thrown: bind: Address already in use 0x00007f7fd0fbb4c6 libUE4Editor-Carla.so!void clmdep_asio::detail::throw_exception<std::__1::system_error>(std::__1::system_error const&)(+0x5) 0x00007f7fd0f792bc libUE4Editor-Carla.so!clmdep_asio::detail::do_throw_`

What does this mean? Has someone encountered this before?

corkyw10, 2021-03-03:
@philippwulff Are you still experiencing this issue? If so can you tell me what version of CARLA you are using?

philippwulff, 2021-03-03:
I am not working on this anymore and since it is so far back I cannot reproduce it

closed 2021-03-03 by philippwulff

emanef13, 2023-05-21:
I faced the same crash error when I was trying to play the simulation (click the play button).
In my case the error appears when another crash has happened before and I clicked "Send and restart".
Instead if you click "Send and close", and relaunch CARLA manually (make launch), then you can normally play the simulation without crashing due to "Exception thrown: bind: Address already in use".

Hope that helps.

torpedlars, 2023-06-30:
Having this issue as well, when pressing play in UE it crashes.
Gives the fatal error : carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.cpp] [Line: 136] Exception thrown: bind: Address already in use << callstack too long >>
Using UE4 4.26 and carla 0.9.14
Not sure what causes it but a restart of the computer removes the error for a while then it comes back

bernatx, 2023-06-30:
Hi,
make sure you don't have any other instance of Carla or other application that is using the same ports (by default, 2000, 2001, 2002). The error is saying that one of those ports are not available.

Also, you can try to use other ports instead of the default ones if required (see commandline parameters like -carla-rpc-port=3000, for example).

Regards,
Bernat

torpedlars, 2023-07-03:
Thank you Bernat for pointing me in the right direction. I will look into that!
Friendly regards
Marcus