carla-simulator/carla #3561: Is there a way to control the path of the NPC vehicle？

lz1159435992, 2020-11-10:
Is there a method to control the path of the NPC vehicle, set the starting and ending coordinates, and drive the NPC vehicle at the specified speed? I see that Agent can do this，but I can‘t find any guide in the new release，is there a new method to do this now？

jackbart94, 2020-11-10:
Hi,

It is possible using ``automatic_control.py`` and the Behavior agent, but I don't think it's well documented. You might have to look into the code, it should be enough to check the flags of the script.

lz1159435992, 2020-11-10:
Thank you！ I find that agent in PythonAPI can control the path but I can‘t find some guide for this. Also It's kind of diffcult for me to modify the `automatic_control.py`. I just want to get some document to guide me.

jackbart94, 2020-11-10:
```        else:
            agent = BehaviorAgent(world.player, behavior=args.behavior)

            spawn_points = world.map.get_spawn_points()
            random.shuffle(spawn_points)

            if spawn_points[0].location != agent.vehicle.get_location():
                destination = spawn_points[0].location
            else:
                destination = spawn_points[1].location

            agent.set_destination(agent.vehicle.get_location(), destination, clean=True)
```

I'm pretty sure it's enough to change the value of ``destination`` to where you want the vehicle to arrive.

```        while self.player is None:
            if not self.map.get_spawn_points():
                print('There are no spawn points available in your map/town.')
                print('Please add some Vehicle Spawn Point to your UE4 scene.')
                sys.exit(1)
            spawn_points = self.map.get_spawn_points()
            spawn_point = random.choice(spawn_points) if spawn_points else carla.Transform()
            self.player = self.world.try_spawn_actor(blueprint, spawn_point)
```

And here if you change the value of ``spawn_point`` you can choose where the vehicle will be created.

lz1159435992, 2020-11-11:
Thank you so much ! I will try this!

lz1159435992, 2020-11-11:
How can I import ` BehaviorAgent` into my code? I have problem to do this. My code can only run in `/PythonAPI/examples` now.  I don't know how to use ` BehaviorAgent`.

jackbart94, 2020-11-11:
``from agents.navigation.behavior_agent import BehaviorAgent``

I am closing this issue since it's not an issue. Check the code of ``automatic_control.py`` further to understand how to run your own code. 

If you need more help feel free to contact us on Discord at [this link.](https://discord.gg/8kqACuC)

closed 2020-11-11 by jackbart94

lz1159435992, 2020-11-11:
ok, thank you so much!

DokurOmkar, 2021-07-14:
@jackbart94 I tried changing the spawn point using 0.9.11 but it does not work

            if not self.map.get_spawn_points():
                print('There are no spawn points available in your map/town.')
                print('Please add some Vehicle Spawn Point to your UE4 scene.')
                sys.exit(1)
            spawn_points = self.map.get_spawn_points()
            current_location = carla.Location(x=-2.419356, y=204.005676, z=0.001191)
            spawn_point = carla.Transform(current_location, carla.Rotation())
            self.player = self.world.try_spawn_actor(blueprint, spawn_point)

I tried the above code and it just stays in the while loop and never comes out of it