carla-simulator/carla #3567: Why the coordinate system of Carla is not same to the real world ?

Zhenyu1, 2020-11-10:
When i use the Carla, I found that when the car is heading **north** and moving forward, the odometer reading **X** increases and the GNSS reading **longitude** increases. Is this normal?

And when the car is heading **west** and moving forward, the odometer reading **Y** increases and the GNSS reading **latitude** increases.

How to solve this problem, let him return to the coordinate system of real world (**north=y, est=x, up=z**).

Zhenyu1, 2020-11-11:
It's just that the car is facing the wrong way.

closed 2020-11-11 by Zhenyu1

Zhenyu1, 2021-10-14:
there is long time, I can't recall it, but it may be related to the **initial pose** of the car.

werewolfdev, 2021-10-15:
Iam actually trying to find a coordinate transformation from the exact real world to the coordinate system in CARLA .Do you have any inisghts on this .Maybe i can make use of openSTREET maps and start both the ego vehicle and exact vehicle at same geodetic coordinates but Iam not sure if thats the right way to do it

Zhenyu1, 2021-10-15:
If i remember correctly, **coordinate system in CARLA** is **UTM coordinate** 

a site for online Transform, http://rcn.montana.edu/Resources/Converter.aspx

hope this helps

AOOOOOA, 2022-05-19:
> @Zhenyu1 Thanks for that . But actually i will receive the real world coordinates from a real vehicle and then I have to simulate scenarios in CARLA and send the relative position of target vehicle back to the vehicle bypassing its real sensors and to trigger the functions. So I receive the real vehicle data in ENU or geodetic to CARLA , but dont know how to use this data along with CARLA coordinates. Thats what confuses me

Hi, have you solved this problem?

werewolfdev, 2022-05-19:
Hey @AOOOOOA . Yes for me ,the issue is taken care of . In my case, the position data received from the real ego  was converted to tangential coordinate systems like ENU (so basically we have x,y , yaw etc). In CARLA we use ESU , which is also tanegntial plane. Hence I used the relative changes in x and y coordinate system to localize the ego vehicle in CARLA with these delta changes in each frame . It works.

2657666247, 2023-07-18:
> 嘿嘿@AOOOOOA是的，对我来说，这个问题已经得到解决。就我而言，从真实接收自我到的位置数据被转换为像ENU这样的切向坐标系（所以基本上我们有x，y，偏在CARLA中，我们使用ESU，因此它也是切面。，我使用x和y坐标系的相对变化来定位CARLA中的自我载具，并在每个帧中使用这些增量变化。有用。

Hello, I want to connect the real vehicle data to carla recently to realize the interaction between the real car and the virtual car. How did you realize it at that time? Can you give me some suggestions or codes, thank you very much.