carla-simulator/carla #3570: Traffic manager gives rpc::timeout while calling RPC function 'unregister_vehicle'

yshen47, 2020-11-10:
Hello, there. Thanks so much, first for offering this brilliant work, which helps my research a lot. 

I am now working towards having multiple training agents running in parallel in no_rendering mode. My program is able to run for about 10 min, then the client will complain rpc::timeout: Timeout of 2000ms while calling RPC function 'unregister_vehicle', I guess it's one function in the source code of traffic manager? And at the end, it just complains about too many open files in a process. 

I wonder why this occurs, and how I should handle this. I am currently running in the latest version of 0.9.10.1. Thanks

jackbart94, 2020-11-11:
Hi,

I am assigning also @pravinblaze to this issue. Before we dig in deeper for a possible bug, I want to make sure of a few things:
- Have you set up the experiment in a way that is compatible with the  [documentation about multiple Traffic Managers?](https://carla.readthedocs.io/en/latest/adv_traffic_manager/#running-multiple-traffic-managers)
- Could you provide more information about the script? Is it running multiple agents in the same CARLA simulation, or do you have more simulations running?
- Are you sure that you are correctly removing the current hero car? You can check how to do so in the several sample scripts we provide.

yshen47, 2020-11-12:
Sorry for the late reply. I managed to circumvent this issue by wrapping the client inside another process. Thanks still!

closed 2020-11-12 by yshen47