carla-simulator/carla #3573: Crash in carla 0.9.10

harmeeksinghbedi, 2020-11-11:
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `python3 Agent_auto_sac_vae_64.py -n 64_sac_retrain'.
Program terminated with signal SIGABRT, Aborted.
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
[Current thread is 1 (Thread 0x7fb49541e700 (LWP 20857))]
(gdb) where
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007fb6548f78b1 in __GI_abort () at abort.c:79
#2  0x00007fb654940907 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7fb654a6ddfa "%s\n")
    at ../sysdeps/posix/libc_fatal.c:181
#3  0x00007fb65494797a in malloc_printerr (str=str@entry=0x7fb654a6fa88 "double free or corruption (fasttop)")
    at malloc.c:5350
#4  0x00007fb65494f074 in _int_free (have_lock=0, p=0x7fb42001d320, av=0x7fb420000020) at malloc.c:4230
#5  __GI___libc_free (mem=0x7fb42001d330) at malloc.c:3124
#6  0x00007fb58814f5b5 in __gnu_cxx::new_allocator<std::__detail::_Hash_node<unsigned long, false> >::deallocate (__p=<optimized out>, this=<optimized out>)
    at /usr/bin/../lib/gcc/x86_64-linux-gnu/7.5.0/../../../../include/c++/7.5.0/ext/new_allocator.h:125
#7  std::allocator_traits<std::allocator<std::__detail::_Hash_node<unsigned long, false> > >::deallocate (
    __p=<optimized out>, __n=1, __a=...)
    at /usr/bin/../lib/gcc/x86_64-linux-gnu/7.5.0/../../../../include/c++/7.5.0/bits/alloc_traits.h:462
#8  std::__detail::_Hashtable_alloc<std::allocator<std::__detail::_Hash_node<unsigned long, false> > >::_M_deallocate_node (this=<optimized out>, __n=<optimized out>)
    at /usr/bin/../lib/gcc/x86_64-linux-gnu/7.5.0/../../../../include/c++/7.5.0/bits/hashtable_policy.h:2086
#9  std::_Hashtable<unsigned long, unsigned long, std::allocator<unsigned long>, std::__detail::_Identity, std::equal_to<unsigned long>, std::hash<unsigned long>, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<false, true, true> >::_M_erase (this=<optimized out>, __bkt=<optimized out>, __prev_n=<optimized out>, __n=<optimized out>)
    at /usr/bin/../lib/gcc/x86_64-linux-gnu/7.5.0/../../../../include/c++/7.5.0/bits/hashtable.h:1890
#10 std::_Hashtable<unsigned long, unsigned long, std::allocator<unsigned long>, std::__detail::_Identity, std::equal_to<unsigned long>, std::hash<unsigned long>, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<false, true, true> >::_M_erase (this=0x7fb4200151a0, __k=<optimized out>)
    at /usr/bin/../lib/gcc/x86_64-linux-gnu/7.5.0/../../../../include/c++/7.5.0/bits/hashtable.h:1916
#11 std::_Hashtable<unsigned long, unsigned long, std::allocator<unsigned long>, std::__detail::_Identity, std::equal_to<unsigned long>, std::hash<unsigned long>, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<false, true, true> >::erase
    (this=0x7fb4200151a0, __k=<optimized out>)
    at /usr/bin/../lib/gcc/x86_64-linux-gnu/7.5.0/../../../../include/c++/7.5.0/bits/hashtable.h:759
#12 std::unordered_set<unsigned long, std::hash<unsigned long>, std::equal_to<unsigned long>, std::allocator<unsigned long> >::erase (this=0x7fb4200151a0, __x=<optimized out>)
    at /usr/bin/../lib/gcc/x86_64-linux-gnu/7.5.0/../../../../include/c++/7.5.0/bits/unordered_set.h:544
#13 carla::traffic_manager::TrackTraffic::RemovePassingVehicle (this=0x55ca4a5d4e00, 
    waypoint_id=3126815332488097760, actor_id=<optimized out>)
    at ../../LibCarla/source/carla/trafficmanager/TrackTraffic.cpp:147
#14 0x00007fb58814dc70 in carla::traffic_manager::TrackTraffic::DeleteActor (this=0x55ca4a5d4e00, actor_id=0)
    at ../../LibCarla/source/carla/trafficmanager/TrackTraffic.cpp:110
#15 0x00007fb58814d1d0 in carla::traffic_manager::TrackTraffic::UpdateUnregisteredGridPosition (
    this=0x55ca4a5d4e00, actor_id=1480, waypoints=std::vector of length 3, capacity 3 = {...})
    at ../../LibCarla/source/carla/trafficmanager/TrackTraffic.cpp:17
#16 0x00007fb58810fe2d in carla::traffic_manager::ALSM::UpdateUnregisteredActorsData (this=0x55ca4a5d5868)
    at ../../LibCarla/source/carla/trafficmanager/ALSM.cpp:309
#17 0x00007fb58810d8d8 in carla::traffic_manager::ALSM::Update (this=<optimized out>)
    at ../../LibCarla/source/carla/trafficmanager/ALSM.cpp:123
#18 0x00007fb5880e12bf in carla::traffic_manager::TrafficManagerLocal::Run (this=0x55ca4a5d4c80)
    at ../../LibCarla/source/carla/trafficmanager/TrafficManagerLocal.cpp:150
#19 0x00007fb64ef466df in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#20 0x00007fb654caf6db in start_thread (arg=0x7fb49541e700) at pthread_create.c:463
---Type <return> to continue, or q <return> to quit---

harmeeksinghbedi, 2020-11-11:
I am using Carla to train a reinforcement model and keep seeing this crash regularly.

harmeeksinghbedi, 2020-11-11:
I am using the following params 
params = {
    'number_of_vehicles': 0,
    'number_of_walkers': 0,
    'display_size': 224,  # screen size of bird-eye render
    'max_past_step': 1,  # the number of past steps to draw
    'dt': 0.1,  # time interval between two frames
    'discrete': False,  # whether to use discrete control space
    'discrete_acc': [0.5, 1.0],                  # discrete value of accelerations
    'discrete_steer': [-1.0, 0.0, 1.0],          # discrete value of steering angles
    'continuous_accel_range': [MIN_SPEED, MAX_SPEED],  # continuous acceleration range
    'continuous_steer_range': [MIN_STEER, MAX_STEER],       # continuous steering angle range
    'ego_vehicle_filter': 'vehicle.lincoln*',    # filter for defining ego vehicle
    'port': 2000,  # connection port
    'town': 'Town03',  # which town to simulate
    'task_mode': 'random',  # mode of the task, [random, roundabout (only for Town03)]
    'max_time_episode': 5000,  # maximum timesteps per episode
    'image_height': 224,
    'image_width': 224,
    'max_waypt': 12,  # maximum number of waypoints
    'obs_range': 32,  # observation range (meter)
    'lidar_bin': 0.125,  # bin size of lidar sensor (meter)
    'd_behind': 12,  # distance behind the ego vehicle (meter)
    'out_lane_thres': 2.0,  # threshold for out of lane
    'desired_speed': 10,  # desired speed (m/s)
    'max_ego_spawn_times': 200,  # maximum times to spawn ego vehicle
    'display_route': True,  # whether to render the desired route
    'pixor_size': 64,  # size of the pixor labels
    'pixor': False,  # whether to output PIXOR observation
}

corkyw10, 2021-03-03:
@harmeeksinghbedi Are you still facing this issue? If so could you give some more context on the issue? Does the error happen under a certain set of circumstances? What version of CARLA are you using?

@jackbart94 Could you follow up please?

jackbart94, 2021-03-03:
Unfortunately, without any way to reproduce this, I can't debug it. We would need more information.

closed 2021-11-18 by jackbart94