carla-simulator/carla #3575: Carla's Automatic Control Agent weird behavior with Synchronization enabled

kochlisGit, 2020-11-11:
Hello everyone, 
I am trying to run an automatic car in synchronization mode, using the agents from carla.agents module. I've tested the agent enabling synchronization and It is working perfectly. However, When I turn on synchronization, The vehicle behavior is unstable, it steers left and right, gets out lane, etc.

        # Enabling syncing....
        sync_settings = world.get_settings()
        sync_settings.fixed_delta_seconds = 0.1
        sync_settings.synchronous_mode = True
        world.apply_settings(sync_settings)

        # Initializing the agent....
        world.tick()
        agent = RoamingAgent(vehicle=vehicle, target_speed=60.0)
        agent.set_destination(dest_location)

        # Getting the agen to run. However, when syncing is enabled, the agent's behavior is unstable.
        while not agent.done():
            world.tick()
            control = agent.run_step(debug=False)
            vehicle.apply_control(control)

However, If i turn off the synchronization, The agent does a perfect job navigation the vehicle.
Does anybody know what is going on with the world.tick() and the Roaming Agent behavior?
Thanks a lot.

FiodarKazhamiaka, 2020-11-11:
I've encounter this same problem in the past. The issue is that simulating at 10 FPS (fixed_delta_seconds = 0.1) is too low for the autopilot, which I've been told is configured to run at 20 FPS. You end up with agents over-correcting turns because each action is applied for 0.1 seconds before it can be re-evaluated.

kochlisGit, 2020-11-12:
@FiodarKazhamiaka 
Indeed, this seems to be the case. I've tried changing the deta time from 0.1 seconds to 0.05 and the agent is now stable.
Thanks a lot man..!

closed 2020-11-12 by kochlisGit

guntherkrehl, 2020-11-12:
Hi there,

I am encountering the same problem with the default autopilot. I have a refresh rate of 0.01s (100Hz), but my simulation is much slower than real time because of all the sensors I simulate. I do run my sim in synchronous mode. With that kind of setup I do still get a similarly weird agent behavior. Resulting in: swerving outside of the lane, sometimes even hitting a barrier (e.g. in Town05), overcorrecting in turns etc. Interestingly enough, this only applies to the ego vehicle with all them sensors attached, not to any other traffic participants, as far as I can see.

Is this another issue unrelated to this one and I should create a new issue? Or can the author open this issue again?

Thanks a bunch!

kochlisGit, 2020-11-12:
First of all, it is important to use the world.tick() command  before you initialize the agent. It is also important to use lower delta time. Try different values from range 0.01 to 0.1 and see how the agent behaves.