carla-simulator/carla #3579: Too much mouse sensibility in server window

davidtena98, 2020-11-12:
Hi,

When running CARLA server, if I try to use the mouse to drag and change the orientation of the camera (not the position), it will, but so much I have no control on it. Is there any way to change mouse sensibility? I don't remember having this issue in previous versions of CARLA

Axel1092, 2021-03-03:
Hi @davidtena98, sorry for the late response. We don't have means to change the sensitivity of the server camera. The rotation will be larger the greater the fps you get at any given point.

uebian, 2021-07-18:
Is there any options to set the sensitivity of the server camera?

satyamjay-iitd, 2022-02-04:
I have the same issue. I am running on ubuntu 18.04 with -vulkan. Mouse sensitivity is too high to navigate around the world. W/A/S/D works fine though. Was anyone able to fix this?

obadul024, 2022-10-18:
having the same issue with version 0.9.10.1
mouse control is practically useless
please open up the issue again
thanks

JiayinDeng, 2022-11-21:
--environment:
carla version 0.9.12

I can adjust the mouse sensitivity in `Editor Preferences`
"Edit"-"Editor Preferences", then search "mouse".
However, moving with the mouse still seems incorrect.

pauln2k, 2023-01-07:
Has anyone solved this issue yet? I am also facing this problem and cant find any type of solution.

swrko, 2023-02-25:
I fixed navigation in my carla server window by adjusting sensitivity parameters for mouse. 
i am using CARLA 0.9.13 installed version as debian package
changes are made in file: /opt/carla-simulator/CarlaUE4/Config/DefaultInput.ini

i recommend you to create backup of the config file before any change, because i dont really know if it can cause some other problems. Hope it will work for you.

i adjusted some lines in the config file :>

```
-AxisConfig=(AxisKeyName="MouseX",AxisProperties=(DeadZone=0.00000,Exponent=1.00000,Sensitivity=0.300,Scale=-1.0000))
-AxisConfig=(AxisKeyName="MouseY",AxisProperties=(DeadZone=0.00000,Exponent=1.00000,Sensitivity=0.300,Scale=-1.0000))
+AxisConfig=(AxisKeyName="MouseX",AxisProperties= (DeadZone=0.000000,Sensitivity=0.3000,Scale=1.0000,Exponent=1.000000,bInvert=False))
+AxisConfig=(AxisKeyName="MouseY",AxisProperties=(DeadZone=0.000000,Sensitivity=0.3000,Scale=1.0000,Exponent=1.000000,bInvert=False))
bUseMouseForTouch=True
bEnableMouseSmoothing=False
```

DanyThodms, 2023-05-31:
We encountered the same problem and ultimately found out that it was because we used a remote connection that caused a mouse mapping problem, which would not have occurred on the computer where the project was located.

GabrielShimanuki, 2023-06-19:
I'm having the same issue. Unfortunately, I can't access the server where the project is running. Do you know how to solve this problem of mouse mapping?
I'm using NoMachine to remote access.

DanyThodms, 2023-06-27:
@GabrielShimanuki Sorry, I haven't completely solved this problem yet.

ajayy189, 2024-04-16:
./CarlaUE4.sh -quality-level=Epic -benchmark -fps=20 -carla-server -windowed -ResX=800 -ResY=600

this command to start carla fixed the mouse sensitivity and FPS issues

Alobaali12, 2024-11-13:
> I fixed navigation in my carla server window by adjusting sensitivity parameters for mouse. i am using CARLA 0.9.13 installed version as debian package changes are made in file: /opt/carla-simulator/CarlaUE4/Config/DefaultInput.ini
> 
> i recommend you to create backup of the config file before any change, because i dont really know if it can cause some other problems. Hope it will work for you.
> 
> i adjusted some lines in the config file :>
> 
> ```
> -AxisConfig=(AxisKeyName="MouseX",AxisProperties=(DeadZone=0.00000,Exponent=1.00000,Sensitivity=0.300,Scale=-1.0000))
> -AxisConfig=(AxisKeyName="MouseY",AxisProperties=(DeadZone=0.00000,Exponent=1.00000,Sensitivity=0.300,Scale=-1.0000))
> +AxisConfig=(AxisKeyName="MouseX",AxisProperties= (DeadZone=0.000000,Sensitivity=0.3000,Scale=1.0000,Exponent=1.000000,bInvert=False))
> +AxisConfig=(AxisKeyName="MouseY",AxisProperties=(DeadZone=0.000000,Sensitivity=0.3000,Scale=1.0000,Exponent=1.000000,bInvert=False))
> bUseMouseForTouch=True
> bEnableMouseSmoothing=False
> ```
> 
> Then with mouse left-click and keys (W,A,S,D) iam able to navigate through the world.

That worked for me only had to change this line from False to True: 

bUseMouseForTouch=True

closed 2026-11-13 by carla-triage[bot]