carla-simulator/carla #3580: Adding Custom Blueprints to the Actor Registry?

VictorWeidar, 2020-11-12:
Hello!

For a while now I have been trying to register new actors to Carla, so that I may access them in our python scripts. After little bit of reverse engineering I managed to figure out how to register classes that were already included in UE4 such as Directional lights or Atmospheric Fog. I did this using the "ActorView.h" and "ActorRegistry.cpp" scripts.

However, when I try to add my own blueprint classes, I just get an error that says "Use of undeclared Identifier" when I try to start the engine. I saw that you had added a "TrafficSignBase" blueprint that derives from AActor and that works fine, however when I try to add my "CustomBlueprintBase" that also derives from AActor I get the error mentioned above.

The error mentioned occurs in the "ActorRegistry.cpp" script first in the part where you cast to the actor type itself. As mentioned this works perfectly when adding things that are included in UE4, but not with my own items.

Do you have any tips that could help me include our own blueprint classes for later use in python? Am I going about the right way to register my own blueprint classes or am I missing a step?

Thanks

corwinjs, 2021-02-16:
Hello,

I've been trying to add and register new actors too. I was able to add my own function that all actors can call using the python API, which I did mostly to learn better how the connection between the python API and the underlying Unreal C++ code works. Using that experience and trying to emulate how something like a Walker was implemented, I've been trying to add my own actor which at first will just have a void function that ends up calling a UE_LOG statement just to verify it's working. I don't have anything to show for it though yet. Is there any documentation for adding an actor to Carla so that it can be spawned via the python API? If not, is that something that someone can help with here?

Thanks

corkyw10, 2021-03-03:
We have the following [tutorial](https://carla.readthedocs.io/en/latest/tuto_D_create_sensor) which may help.

Otherwise, @bernatx do you have any advice on this? Maybe we can make some changes and updates to the docs

corwinjs, 2021-03-16:
Hi Corrine,

That tutorial has been very helpful. It is not what I was looking for originally, but it has helped me create a workaround.

I was hoping to be able to add and register entirely new actors in some way, with zero or very minimal code changes to those actors. Instead, my workaround has been to create what I call "interface actors", which are actually just sensors that don't have serializers, created using that tutorial as a starting point. I then added functions to an interface actor to spawn and control the unaltered actor that I was trying to add to Carla in the first place. The trickiest part from there was learning how to hook up those function to the Python API so that a user can actually spawn and interact with my actors (via their interface actors). [This response](https://github.com/carla-simulator/carla/issues/1615#issuecomment-536386582) was very useful as a starting point for the process of adding those functions.

Thanks for your help!
-Corwin

Weidvic, 2021-06-21:
Thanks for all of your responses, and I apologize for not being able to get to them sooner. I wanted to try using the sensor tutorial for assistance, but i keep getting an error when using "make rebuild" at the end of the tutorial.

the rebuild process breaks at SafeDistanceEvent.h, where it gives me the following error:

```
../../LibCarla/cmake/../source/carla/sensor/data/SafeDistanceEvent.h:20:8: error: no matching constructor for initialization of 'Array<rpc::ActorId>' (aka 'Array<unsigned int>')
     : Array<rpc::ActorId>(std::move(data)) {}
       ^                   ~~~~~~~~~~~~~~~
../../LibCarla/cmake/../source/carla/sensor/data/Array.h:23:9: note: candidate constructor (the implicit copy constructor) not viable: no known conversion from 'typename std::remove_reference<RawData &>::type' (aka 'carla::sensor::RawData') to 'const carla::sensor::data::Array<unsigned int>' for 1st argument
  class Array : public SensorData {
        ^
../../LibCarla/cmake/../source/carla/sensor/data/Array.h:127:14: note: candidate constructor template not viable: requires 2 arguments, but 1 was provided
    explicit Array(RawData &&data, FuncT get_offset)
             ^
../../LibCarla/cmake/../source/carla/sensor/data/Array.h:136:14: note: candidate constructor not viable: requires 2 arguments, but 1 was provided
    explicit Array(size_t offset, RawData &&data)
             ^
```

What could be the cause of this @corkyw10 ? I tried this both on Carla versions 0.9.10 and 0.9.11 and got the same errors.

MaikolDrechsler, 2021-12-06:
Hello VicWei, 

I am facing exactly the same problem. I am using the Carla Version 0.9.12. 
Did you find a solution? 

Thank you in advance, 
Maikol

Weidvic, 2021-12-06:
Hello,

As this was over a year ago and i no longer work with the same company I did then, I have not persued a solution. It's a C++ / Unreal Engine error, that's as much as I know. 

I hope you find the solution, and I apologize for not being able to help further.

Best regards,
Vic

ronyshaji, 2022-11-24:
@corwinjs Hello, Are you able to find a solution ? I am looking for the solution and stuck with it. Any comments will be helpful.

corwinjs, 2022-12-04:
@Weidvic @MaikolDrechsler @ronyshaji In case anyone is still looking for the solution to those immediate build problems, the tutorial has a few mistakes and these are the fixes:
- LibCarla/source/carla/sensor/data/SafeDistanceEvent.h: Change `Array<rpc::ActorId>(std::move(data)) {}` to `Array<rpc::ActorId>(0u, std::move(data)) {}`
- LibCarla/source/carla/sensor/s11n/SafeDistanceSerializer.h: Change `ActorId id = episode.FindActor(actor).GetActorId()` to `ActorId id = episode.FindCarlaActor(actor)->GetActorId()`
- Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/Sensor.h: Change `void Tick(const float DeltaTime) final` to `void Tick(const float DeltaTime)`

Credit largely to those who posted about these fixes in the source code gist linked to [here](https://gist.github.com/nsubiron/011fd1b9767cd441b1d8467dc11e00f9) and at the top of the tutorial

closed 2024-05-10 by MarcelPiNacy-CVC