carla-simulator/carla #3583: Carla Remote Connection Display

FarukOzderim, 2020-11-13:
Hello guys, did any of you successfully run Carla with remote connection?

I can run Carla without display and test examples, however I can not run Carla with display. I use aws instance with nomachine remote connection. Here are my results:
I think I take opengl errors, but my video driver should support opengl..

[image]

walzimmer, 2020-11-17:
Hi Faruk, 

seems like you have OpenGL 3.1 installed. Try running it with the -opengl3 flag:

CarlaUE4.sh -opengl3

Best,
Walter

FarukOzderim, 2020-11-18:
Hello @walzimmer , I have used -opengl3 flag in the second image.

walzimmer, 2020-11-22:
Then install OpenGL 4 and run it with the `-opengl` flag:
`$sudo apt install mesa-utils`

grvbhosale, 2020-11-23:
Did you get it to work ? I am facing the same problem

FarukOzderim, 2020-11-23:
No I haven't. It seems there is no one who solved this issue.

germanros1987, 2021-03-03:
Hi @FarukOzderim . I do run CARLA remotely via VNC. There are a few things to consider in order to do the right setup.

First, you need to run a X.org server in the remote node. 
On top of that you need an X-zero VNC server. This means a VNC server that doesn't bring its own X server. A good option is tigerVNC. The command to run the X0VNC server typically looks like this: 
```x0vncserver -Display=:0 -NeverShared -PollingCycle=75 -rfbauth=your_vnc_folder/.vnc/passwd```

This assumes you have an Xorg server running at :0

If you are planning on running the default graphics backend of CARLA (i.e. Vulkan). You would need to have the latest NVIDIA driver installed in the remote node, along with libvulkan.

Best regards.

closed 2023-06-02 by stale[bot]