carla-simulator/carla #3585: Help with plotting

venkatramanrenganathan, 2020-11-13:
Hi All,

I would like to get some help with my Carla plotting. I would love to get an ellipse under my car to show my car uncertainty. And I would also love to plot my car trajectories like this. Please help me. PS: the carla draw_line function is not plotting at arbitrary points. It is always plotting either on the lane mid point or exactly on the white lane separator. I want it to plot my trajectory starting at any arbitrary point in lane and end similarly at another point. Please help me. A small guidance as to how to achieve this would help me a lot please. I made the following plot using paint !!!

 I am using Windows 10 machine with CARLA 0.9.10 latest code. 

Warm Regards,
Venkatraman Renganathan,

[image]

goksanisil23, 2020-11-18:
Hi, 
Which version of Carla release are you using and could you share the code snippet where you use draw_line.

/Göksan

venkatramanrenganathan, 2020-11-18:
Hi,

This is my plotting code. It is not plotting at arbitrary points. Only in the center of lane or on the divider line. I made the above figure using paint depicting the feature that i want. I downloaded the CARLA as soon as 0.9.10 got released.  
self.debug.draw_line(start_waypoint.location + carla.Location(z=0.25), 
                                  end_waypoint.location + carla.Location(z=0.25), 
                                  thickness=0.1, color=red, life_time=20)

corkyw10, 2021-03-09:
Hi @venkatramanrenganathan instead of the elipse, you could use the built in [DebugHelper.draw_box](https://carla.readthedocs.io/en/latest/python_api/#carladebughelper) to draw the bounding box of the car. I also suggest you have a look at the snippet for [DebugHelper.draw_string](https://carla.readthedocs.io/en/latest/python_api/#methods_11) combined with the [lane_explorer.py](https://github.com/carla-simulator/carla/blob/master/PythonAPI/util/lane_explorer.py) script, there are some useful functions defined there that could help you.

closed 2021-08-05 by corkyw10

SaeedBASALEH, 2023-03-27:
I have a similer problem. I want to draw trajectories which is not in a line form. Basically, I want to draw trajectories with curvature . Which carla function would be more adapted for doning that? 

Thank you !