carla-simulator/carla #3586: Imposing wheel torque in Carla Vehicle

brscholz, 2020-11-15:
Hi everyone,

I'm trying to couple CARLA with a vehicle powertrain model in Simulink in the following way:

1.     Accelerator and brake pedal position come from an external source and are fed into the Simulink model.
2.     The Simulink model does the torque calculation from engine to wheel input.
3.     CARLA receives the wheel torque and does the calculation of acceleration force and vehicle speed.
4.     CARLA vehicle speed is fed into the Simulink model for calculation of the resulting engine speed.

Is there a way yet to directly set the torque that is applied to each wheel?

A workaround I thought about could be the actor.set_force API function, but that way I would lose effects of two- and four-wheel-drives and many more.

Thanks in advance and best regards
Robert

brscholz, 2020-11-19:
Good morning,

I have found out that my problem would be solved if I had access to USimpleWheeledVehicleMovementComponent in Carla, which orginally uses UWheeledVehicleMovementComponent. Maybe someone hat the same problem and already did the job to implement CarlaSimpleWheeledVehicle?

https://docs.unrealengine.com/en-US/API/Plugins/PhysXVehicles/USimpleWheeledVehicleMovementCom-/index.html

Best regards
Robert

DSantosO, 2021-03-10:
Hello @brscholz,
In PR #3868, we are adding control over the rotation of the wheel from the client, could this feature help you?

aufkeinenfallvergessen, 2022-02-28:
Hello @brscholz,
did you find a solution for your Problem?
I have a similar problem. I want to brake the wheels individually.

@DSantosO is there a way to add the Function "SetBrakeTorque()" from "USimpleWheeledVehicleMovementComponent" as well?
Please have a look to my post regarding my problem:  [#5215](https://github.com/carla-simulator/carla/discussions/5215)

brscholz, 2022-02-28:
Hi @aufkeinenfallvergessen, I decided to use a different approach to my problem back then, sorry. Still, adding a class ACarla**Simple**WheeledVehicle would be nice - or better an option to turn the ACarlaWheeledVehicle into a SimpleWheeledVehicle. The latter way the current vehicle models would still be usable I guess.

brscholz, 2022-03-01:
Hey, we reduced the effort because it was only an internal showcase. What we ended up with was calculating a corrected throttle signal so we would get the exact same engine torque output as in our simulink model. That won't be helping you with setting each wheel's torque, I'm afraid.

aufkeinenfallvergessen, 2022-03-01:
Thx for letting me know :)
Yeah unfortunately it doesnt help me, since I need to apply individual brake torques to each wheel at each tick