carla-simulator/carla #3590: Building docker fails with Cook failed error

ptigas, 2020-11-16:
Hi,

Following the instructions from here https://github.com/carla-simulator/carla/tree/dev/Util/Docker fails at the `docker build -t carla -f Carla.Dockerfile .` step.

Here is a tail of logs `/home/ue4/Library/Logs/Unreal Engine/LocalBuildLogs/Log.txt`:

```
ProcessResult.StdOut:   LogTexture: Display: Building textures: T_Male1Skin_d (AutoDXT, 2048X2048)
ProcessResult.StdOut:   LogTexture: Display: Building textures: T_Female1Cloth_v3_d (AutoDXT, 4096X4096)
ProcessResult.StdOut:   LogTexture: Display: Building textures: T_Female1Cloth_v2_d (AutoDXT, 4096X4096)
CommandUtils.Run: Took 9817.150118s to run UE4Editor, ExitCode=137
InternalUtils.SafeDeleteFile: SafeDeleteFile /home/ue4/Library/Logs/Unreal Engine/LocalBuildLogs/Cook-2020.11.16-16.36.38.txt
InternalUtils.SafeCopyFile: SafeCopyFile /home/ue4/UnrealEngine/Engine/Programs/AutomationTool/Saved/Cook-2020.11.16-13.53.00.txt /home/ue4/Library/Logs/Unreal Engine/LocalBuildLogs/Cook-2020.11.16-16.36.38.txt
InternalUtils.SafeDeleteFile: SafeDeleteFile /home/ue4/UnrealEngine/Engine/Programs/AutomationTool/Saved/Cook-2020.11.16-13.53.00.txt
Log.WriteException: ==============================================================================
Log.WriteException: ERROR: Cook failed.
Log.WriteException:        (see /home/ue4/Library/Logs/Unreal Engine/LocalBuildLogs/Log.txt for full exception trace)
Log.WriteException:
Log.WriteException: CommandletException: Editor terminated with exit code 137 (signal 9) while running Cook for /home/ue4/carla/Unreal/CarlaUE4/CarlaUE4.uproject; see log /home/ue4/Library/Logs/Unreal Engine/LocalBuildLogs/Cook-2020.11.16-16.36.38.txt
Log.WriteException:   at AutomationTool.CommandUtils.RunCommandlet (Tools.DotNETCommon.FileReference ProjectName, System.String UE4Exe, System.String Commandlet, System.String Parameters, System.String& DestLogFile) [0x0076d] in <00f0b7c43b86450b9446ac9e7ca71519>:0
Log.WriteException:   at AutomationTool.CommandUtils.RunCommandlet (Tools.DotNETCommon.FileReference ProjectName, System.String UE4Exe, System.String Commandlet, System.String Parameters) [0x00001] in <00f0b7c43b86450b9446ac9e7ca71519>:0
Log.WriteException:   at AutomationTool.CommandUtils.CookCommandlet (Tools.DotNETCommon.FileReference ProjectName, System.String UE4Exe, System.String[] Maps, System.String[] Dirs, System.String InternationalizationPreset, System.String[] CulturesToCook, System.String TargetPlatform, System.String Parameters) [0x0013e] in <00f0b7c43b86450b9446ac9e7ca71519>:0
Log.WriteException:   at Project.Cook (AutomationTool.ProjectParams Params) [0x007ba] in <1306ac9a34ad40329f17e749af162a81>:0 Wrapped by AutomationException: Cook failed.
Log.WriteException:   at Project.Cook (AutomationTool.ProjectParams Params) [0x007f7] in <1306ac9a34ad40329f17e749af162a81>:0
Log.WriteException:   at BuildCookRun.DoBuildCookRun (AutomationTool.ProjectParams Params) [0x00074] in <1306ac9a34ad40329f17e749af162a81>:0
Log.WriteException:   at BuildCookRun.ExecuteBuild () [0x00040] in <1306ac9a34ad40329f17e749af162a81>:0
Log.WriteException:   at AutomationTool.BuildCommand.Execute () [0x00001] in <00f0b7c43b86450b9446ac9e7ca71519>:0
Log.WriteException:   at AutomationTool.Automation.Execute (System.Collections.Generic.List`1[T] CommandsToExecute, System.Collections.Generic.Dictionary`2[TKey,TValue] Commands) [0x0007c] in <00f0b7c43b86450b9446ac9e7ca71519>:0
Log.WriteException:   at AutomationTool.Automation.Process (System.String[] Arguments, UnrealBuildTool.StartupTraceListener StartupListener) [0x00282] in <00f0b7c43b86450b9446ac9e7ca71519>:0
Log.WriteException:   at AutomationTool.Program.MainProc (System.String[] Arguments, UnrealBuildTool.StartupTraceListener StartupListener) [0x00001] in <cac33d5dd49b4eb89cf6aeb78ec98853>:0
Log.WriteException:   at AutomationTool.Program+<>c__DisplayClass1_0.<Main>b__2 () [0x00000] in <cac33d5dd49b4eb89cf6aeb78ec98853>:0
Log.WriteException:   at AutomationTool.InternalUtils.RunSingleInstance (System.Func`1[TResult] Main) [0x000a0] in <00f0b7c43b86450b9446ac9e7ca71519>:0
Log.WriteException:   at AutomationTool.Program.Main (System.String[] Arguments) [0x0020a] in <cac33d5dd49b4eb89cf6aeb78ec98853>:0
Log.WriteException: ==============================================================================
Program.Main: AutomationTool exiting with ExitCode=25 (Error_UnknownCookFailure)
```

and for `/home/ue4/Library/Logs/Unreal Engine/LocalBuildLogs/Cook-2020.11.16-16.36.38.txt`

```
LogTexture: Display: Building textures: M_VariousProps_n (BC5, 2048X2048)
LogCook: Display: Cooked packages 1000 Packages Remain 1792 Total 2792
LogTexture: Display: Building textures: T_LicensePlate_d (AutoDXT, 512X512)
LogCook: Display: Cooked packages 1050 Packages Remain 1792 Total 2842
LogTexture: Display: Building textures: M_VariousProps_ormh (AutoDXT, 2048X2048)
LogMaterial: Display: Missing cached shader map for material M_TrafficLight, compiling.
LogMaterial: Display: Missing cached shader map for material M_TrafficLight, compiling.
```

any clues on why this might be happening? 

Also possibly related to this https://github.com/carla-simulator/carla/issues/3163#issuecomment-675530054

thillRobot, 2020-12-11:
I am having the same problem. I am trying to build carla with docker following the same instructions, and in the last step the cook fails while compiling shaders with `ExitCode=127`.

```
  LogShaderCompilers: Display: Worker (1/5): shaders left to compile 8665
  LogTexture: Display: Building textures: T_v7BPR_ID_B (AutoDXT, 2048X2048)
  LogTexture: Display: Building textures: T_v7BPR_DepthB (AutoDXT, 2048X2048)
  LogTexture: Display: Building textures: T_v7BPR_Mask_B (AutoDXT, 2048X2048)
  LogTexture: Display: Building textures: T_v7BPR_RenderB (AutoDXT, 2048X2048)
  LogTexture: Display: Building textures: T_v7Hair_ROOT_B (AutoDXT, 2048X2048)
  LogTexture: Display: Building textures: T_afro02_BPR_ID (AutoDXT, 2048X2048)
  LogTexture: Display: Building textures: T_afro02_BPR_Mask (AutoDXT, 2048X2048)
  LogTexture: Display: Building textures: T_afro02_BPR_Depth (AutoDXT, 2048X2048)
  LogTexture: Display: Building textures: T_afro02_BPR_Render (AutoDXT, 2048X2048)
  LogTexture: Display: Building textures: T_afro02_BPR_ROOT (AutoDXT, 2048X2048)
  LogTexture: Display: Building textures: T_afroSkin_d (AutoDXT, 4096X4096)
  LogTexture: Display: Building textures: T_v3Skin__orm (AutoDXT, 8192X8192)
Took 1641.562567s to run UE4Editor, ExitCode=137
ERROR: Cook failed.
       (see /home/ue4/Library/Logs/Unreal Engine/LocalBuildLogs/Log.txt for full exception trace)
AutomationTool exiting with ExitCode=25 (Error_UnknownCookFailure)
RunUAT ERROR: AutomationTool was unable to run successfully.
Util/BuildTools/Linux.mk:16: recipe for target 'package' failed
```
I noticed that if I try again it fails at a similar but different point in the cook. For example it also failed on the following shader.

```
LogShaderCompilers: Display: Worker (3/5): shaders left to compile 8799
```

This behavior makes me think the problem may be resource dependent, but my machine meets the described requirements. I tried reading the log that is mentioned in the error, but I do not know enough about the build to decipher the log. I have included part of  `/home/ue4/Library/Logs/Unreal Engine/LocalBuildLogs/Log.txt` below. 

```
ProcessResult.StdOut:   LogShaderCompilers: Display: Worker (1/5): shaders left to compile 8665
ProcessResult.StdOut:   LogTexture: Display: Building textures: T_v7BPR_ID_B (AutoDXT, 2048X2048)
ProcessResult.StdOut:   LogTexture: Display: Building textures: T_v7BPR_DepthB (AutoDXT, 2048X2048)
ProcessResult.StdOut:   LogTexture: Display: Building textures: T_v7BPR_Mask_B (AutoDXT, 2048X2048)
ProcessResult.StdOut:   LogTexture: Display: Building textures: T_v7BPR_RenderB (AutoDXT, 2048X2048)
ProcessResult.StdOut:   LogTexture: Display: Building textures: T_v7Hair_ROOT_B (AutoDXT, 2048X2048)
ProcessResult.StdOut:   LogTexture: Display: Building textures: T_afro02_BPR_ID (AutoDXT, 2048X2048)
ProcessResult.StdOut:   LogTexture: Display: Building textures: T_afro02_BPR_Mask (AutoDXT, 2048X2048)
ProcessResult.StdOut:   LogTexture: Display: Building textures: T_afro02_BPR_Depth (AutoDXT, 2048X2048)
ProcessResult.StdOut:   LogTexture: Display: Building textures: T_afro02_BPR_Render (AutoDXT, 2048X2048)
ProcessResult.StdOut:   LogTexture: Display: Building textures: T_afro02_BPR_ROOT (AutoDXT, 2048X2048)
ProcessResult.StdOut:   LogTexture: Display: Building textures: T_afroSkin_d (AutoDXT, 4096X4096)
ProcessResult.StdOut:   LogTexture: Display: Building textures: T_v3Skin__orm (AutoDXT, 8192X8192)
CommandUtils.Run: Took 1641.562567s to run UE4Editor, ExitCode=137
InternalUtils.SafeDeleteFile: SafeDeleteFile /home/ue4/Library/Logs/Unreal Engine/LocalBuildLogs/Cook-2020.12.11-12.37.20.txt
InternalUtils.SafeCopyFile: SafeCopyFile /home/ue4/UnrealEngine/Engine/Programs/AutomationTool/Saved/Cook-2020.12.11-12.09.59.txt /home/ue4/Library/Logs/Unreal Engine/LocalBuildLogs/Cook-2020.12.11-12.37.20.txt
InternalUtils.SafeDeleteFile: SafeDeleteFile /home/ue4/UnrealEngine/Engine/Programs/AutomationTool/Saved/Cook-2020.12.11-12.09.59.txt
Log.WriteException: ==============================================================================
Log.WriteException: ERROR: Cook failed.
Log.WriteException:        (see /home/ue4/Library/Logs/Unreal Engine/LocalBuildLogs/Log.txt for full exception trace)
Log.WriteException: 
Log.WriteException: CommandletException: Editor terminated with exit code 137 (signal 9) while running Cook for /home/ue4/carla/Unreal/CarlaUE4/CarlaUE4.uproject; see log /home/ue4/Library/Logs/Unreal Engine/LocalBuildLogs/Cook-2020.12.11-12.37.20.txt
```

It seems that several people are having this issue. If someone could figure this out or share some useful information that would be greatly appreciated.

thillRobot, 2020-12-31:
I believe I have solved this issue by increasing the size if the swap partition of my system. I was using the default Ubuntu partitioning with LVM which setup a swap partition of <1G with my 16G of RAM. I increased the swap partition significantly, and now I am not having the error. During the build, `top` shows at least ~12G of RAM and ~4G of swap used at times.

sundw2014, 2021-02-12:
@thillRobot 's solution worked for me. In my case, 32G MEM + 18 G SWP worked, but 32G MEM + 2G SWP did not.

closed 2021-03-03 by corkyw10