carla-simulator/carla #3592: time-out of 10000ms while waiting for the simulator

TeeManiac, 2020-11-17:
I have been working with Carla and scenario runner for past 4 weeks and everything was working fine. I didnt make any changes to carla and was experimenting with scenario runner. For the past few days I'm getting the given error when trying to run scenario runner. I resolve this issue by rebooting the system and then waiting for some time. But this is not a feasible solution to continue with. Please provide support for this issue.

I have referred to #1501  but there is no exact reason or workaround mentioned.

C:\Users\dvk\Documents\scenario Runner\scenario_runner>python scenario_runner.py --openscenario srunner\examples\LaneChangeSimple.xosc
Traceback (most recent call last):
File "C:\Users\dvk\Documents\scenario Runner\scenario_runner\scenario_runner.py", line 598, in sys.exit(main())
File "C:\Users\dvk\Documents\scenario Runner\scenario_runner\scenario_runner.py", line 587, in main scenario_runner = ScenarioRunner(arguments)
File "C:\Users\dvk\Documents\scenario Runner\scenario_runner\scenario_runner.py", line 91, in init self.traffic_manager = self.client.get_trafficmanager(int(self._args.trafficManagerPort))
RuntimeError: time-out of 10000ms while waiting for the simulator, make sure the simulator is ready and connected to 127.0.0.1:2000

**Desktop **

OS: [Windows 10]
CARLA Version [CARLA 0.9.10]
Python version [3.9.0]
Scenario_runner Version [0.9.10]

YenTran32, 2020-12-01:
Hello, I'm struggling with installing carla. Could you please give a more detail on the setting, specifically the versions of Cmake, Make, and Unreal Engine, where you successfully installed carla?

Thank you.

corkyw10, 2021-03-03:
@TeeManiac Thanks for your patience, are you still experiencing this issue?

@YenTran32 if you are installing on windows, there is updated documentation [here](https://carla.readthedocs.io/en/latest/build_windows/#software-requirements) and in linux you'll find the dependencies needed [here](https://carla.readthedocs.io/en/latest/build_linux/#dependencies). If you have any specific issues then the [forum](https://forum.carla.org/) is a good place to ask

YenTran32, 2021-03-04:
Hi Corinne,

Thank you very much for the instructions. I will try to re-compile Carla
from source following the instructions and I will post my result to the
forum later when I come back to this.

However, I think the solution you mentioned is for those who build Carla
from source. In case I use a pre-compiled release of Carla, e.g., just
download from this link
<https://carla-releases.s3.eu-west-3.amazonaws.com/Linux/CARLA_0.9.11.tar.gz>
and run, is there any other solution?

Thanks. Regards.

Tran.

corkyw10, 2021-03-05:
Hi @YenTran32 yes the links I posted are for people who are building CARLA from source. I assumed that was your case because you asked about make, cmake etc. That software is only required for people building from source. 

If you want to use the pre-compiled, packaged version of CARLA then the instructions are found [here](https://carla.readthedocs.io/en/latest/start_quickstart/). You will only need to have installed Python and two Python modules, pygame and numpy.

YenTran32, 2021-03-10:
Hi Corinne,

Thank you for your answer.

Best,

closed 2021-08-05 by corkyw10