carla-simulator/carla #3594: exposure-mode auto - which gamma finally used

mawi42, 2020-11-17:
Question:
You set in Carla 0.9.10 the camera exposure-mode to histogram. From documentation I learned
"Auto exposure Modifies the image gamma to simulate the eye adaptation to darker or brighter areas."
When I render finally the images I read camera attributes, but it seems the gamma value is not changed, even if I see in my scene (no movement static position) that the images are getting darker - so the auto feature seems to be active.
Is there a chance to get current used parameter of the camera like gamma or exposure time? If not, this would be a feature request.

Thanks in advance
Matthias

[image]

corkyw10, 2021-03-04:
@mawi42 thanks for your patience

@DSantosO could you have a look at this please

mawiso, 2021-07-06:
Hello, is there any update on this issue?

closed 2023-06-02 by stale[bot]