carla-simulator/carla #3597: Compiled PythonAPI on macOS fails to load in Python

wbridewell, 2020-11-17:
Carla 0.9.10.1 on macOS 11 (Big Sur) using llvm@8 from homebrew

I was previously able to compile and use the 0.9.9 PythonAPI without any problems (apart from minor changes to the code and setup files). After getting 0.9.10.1 to build, I run into a symbol error when importing carla. The error trace is below. The library for osm2odr is built without issue. Any ideas?

>>> import carla
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/name/carla-0.9.10.1-pythonAPI/carla-0.9.10-py3.9-macosx-10.15-x86_64/carla/__init__.py", line 8, in <module>
    from .libcarla import *
ImportError: dlopen(/Users/name/carla-0.9.10.1-pythonAPI/carla-0.9.10-py3.9-macosx-10.15-x86_64/carla/libcarla.cpython-39-darwin.so, 2): Symbol not found: __ZN7osm2odr21ConvertOSMToOpenDRIVEENSt3__112basic_stringIcNS0_11char_traitsIcEENS0_9allocatorIcEEEENS_15OSM2ODRSettingsE
  Referenced from: /Users/name/carla-0.9.10.1-pythonAPI/carla-0.9.10-py3.9-macosx-10.15-x86_64/carla/libcarla.cpython-39-darwin.so
  Expected in: flat namespace
 in /Users/name/carla-0.9.10.1-pythonAPI/carla-0.9.10-py3.9-macosx-10.15-x86_64/carla/libcarla.cpython-39-darwin.so

gaffordb, 2021-01-24:
I had this problem and found that my `extra_link_args` in `setup.py` was missing the following:
```python
os.path.join(pwd, 'dependencies/lib/libosm2odr.a'),
os.path.join(pwd, 'dependencies/lib/libxerces-c.a'),
```

corkyw10, 2021-03-04:
@wbridewell Thanks for your patience, are you still facing this issue?

wbridewell, 2021-03-04:
I moved forward to 0.9.11 and with this and a few other changes was able to get the Python API working.

closed 2021-03-04 by wbridewell

calix35, 2021-03-08:
> Carla 0.9.10.1 on macOS 11 (Big Sur) using llvm@8 from homebrew
> 
> I was previously able to compile and use the 0.9.9 PythonAPI without any problems (apart from minor changes to the code and setup files). After getting 0.9.10.1 to build, I run into a symbol error when importing carla. The error trace is below. The library for osm2odr is built without issue. Any ideas?
> 
> > > > import carla
> > > > Traceback (most recent call last):
> > > > File "", line 1, in 
> > > > File "/Users/name/carla-0.9.10.1-pythonAPI/carla-0.9.10-py3.9-macosx-10.15-x86_64/carla/**init**.py", line 8, in 
> > > > from .libcarla import *
> > > > ImportError: dlopen(/Users/name/carla-0.9.10.1-pythonAPI/carla-0.9.10-py3.9-macosx-10.15-x86_64/carla/libcarla.cpython-39-darwin.so, 2): Symbol not found: __ZN7osm2odr21ConvertOSMToOpenDRIVEENSt3__112basic_stringIcNS0_11char_traitsIcEENS0_9allocatorIcEEEENS_15OSM2ODRSettingsE
> > > > Referenced from: /Users/name/carla-0.9.10.1-pythonAPI/carla-0.9.10-py3.9-macosx-10.15-x86_64/carla/libcarla.cpython-39-darwin.so
> > > > Expected in: flat namespace
> > > > in /Users/name/carla-0.9.10.1-pythonAPI/carla-0.9.10-py3.9-macosx-10.15-x86_64/carla/libcarla.cpython-39-darwin.so

Do you found the solution for this problem?