carla-simulator/carla #3599: not able to run carla docker

kvaravinda, 2020-11-18:
I am trying to run the Carla simulator as a docker container using the below link.

https://carla.readthedocs.io/en/latest/build_docker/

I have installed docker 19.03, Nvidia drivers 450, and Nvidia-docker2 in an azure Ubuntu 18.04 machine(NV6 series), and it has a 16Gb graphics card(NVIDIA Corporation GM204GL [Tesla M60]).

But when I am trying to create a container with the latest Carla image(carlasim/carla) it is throwing error as given below, it has created a container which binds to 2000-2002 ports but it is not launching any simulator.

the command which I am running - docker run -it -p 2000-2002:2000-2002 --gpus all carlasim/carla ./CarlaUE4.sh

error - "4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
sh: 1: xdg-user-dir: not found"

Am I doing something wrong here or do I need to install anything else. Or is it working correctly and I am not able to realize that?. Please suggest.

ebadi, 2020-11-19:
Duplicate of #1892
I also have seen the error message, I guess it is working correctly and you can ignore the message.

arvindvk555, 2020-11-19:
@ebadi , What does it is "working correctly" mean? bcoz it is either not launching any simulator window or container is up. Container is getting exited immediately as show below. 

simple@carla-linux:~$ docker ps -a
CONTAINER ID        IMAGE                   COMMAND                  CREATED             STATUS                      PORTS               NAMES
6aa7e36bbe62        carlasim/carla:0.9.5    "./CarlaUE4.sh"          4 hours ago         Exited (130) 4 hours ago                        nice_moore
efda5409478e        carlasim/carla:0.9.10   "./CarlaUE4.sh"          4 hours ago         Exited (1) 4 hours ago                          gracious_cerf

I tried to run previous versions (0.9.5) and it has started a container but there was no simulator window launched in this case also. But i can see that the GPU was getting consumed and below mentioned are some of the logs which I was getting when I started the container.

"[2020.11.18-09.41.48:745][ 0]LogAIModule: Creating AISystem for world Town03 [2020.11.18-09.41.49:283][ 0]LogLoad: Game class is 'TheNewCarlaGameMode_C' [2020.11.18-09.41.49:362][ 0]LogWorld: Bringing World /Game/Carla/Maps/Town03.Town03 up for play (max tick rate 0) at 2020.11.18-09.41.49 [2020.11.18-09.41.49:363][ 0]LogCarlaServer: Initialized CarlaServer: Ports(rpc=2000, streaming=2001) [2020.11.18-09.41.49:376][ 0]LogWorld: Bringing up level for play took: 0.091020 [2020.11.18-09.41.49:385][ 0]LogCarlaServer: New episode 'Town03' started [2020.11.18-09.41.49:388][ 0]LogLoad: Took 3.151176 seconds to LoadMap(/Game/Carla/Maps/Town03) [2020.11.18-09.41.50:813][ 0]LogLoad: (Engine Initialization) Total time: 7.31 seconds [2020.11.18-09.41.50:889][ 0]LogRenderer: Reallocating scene render targets to support 856x640 Format 10 NumSamples 1 (Frame:1). [2020.11.18-09.41.50:905][ 0]LogRHI: Using OpenGL program LRU cache: 0 [2020.11.18-09.41.55:968][ 0]LogRenderer: Warning: Gaussian DOF algorithm is deprecated and will be removed from deferred shading renderer. Consider using the Circle DOF method in post process settings. [2020.11.18-09.41.57:860][ 0]LogLinux: Setting swap interval to 'Immediate' [2020.11.18-09.41.57:860][ 0]LogLinux: Warning: Unable to set desired swap interval 'Immediate' [2020.11.18-09.41.57:860][ 0]LogSlate: Slate font cache was flushed [2020.11.18-09.42.57:591][407]LogCarla: Replayer stop [2020.11.18-09.43.58:693][868]LogTaskGraph: Warning: Task graph took 20.03ms for RT to recieve broadcast.

ebadi, 2020-11-19:
It was a while back that I tried Carla docker to have a [**headless** server.](https://carla.readthedocs.io/en/latest/build_docker/). Using docker, I am not sure if you can still have the simulator window.

arvindvk555, 2020-11-19:
when I am running the latest version it is throwing error and exiting the container, there are no other logs. I have tried the suggestions given in the issue which you have tagged above but none of them worked for me.

KunanonxGithub, 2020-11-26:
@ebadi Hello, could you help me throught this question pls? I am so curious about how physic in CARLA actually work because I want to use CARLA and PID control to develop Adaptive Cruise Control and I think it need information such as vehicle mass (which is known) . The problem is that b air coefficient drag force espicially relationship between throttle and velocity of CARLA car. #3625 are not known.  Do you know where can I see those information?

arvindvk555, 2020-11-27:
@ebadi , Hello, When i am running a container with 0.9.5 version it is launcing a container and connection to simulator is also successful as shown below.

COMMAND:  docker run -it -p 2000-2002:2000-2002 --gpus all carlasim/carla:0.9.5 /bin/bash -c './CarlaUE4.sh -nosound -opengl'
RESULT:  <xxx>@<xxx>:~/PythonAPI_0.9.10/util$ python3 test_connection.py 
CARLA 0.9.5 connected at 127.0.0.1:2000.

But when i am running container using 0.9.10 image and trying to test connection to simulator it is not working. Could you please help me out here.

COMMAND: docker run -it -p 2000-2002:2000-2002 --gpus all carlasim/carla:0.9.10 /bin/bash -c 'SDL_VIDEODRIVER=offscreen ./CarlaUE4.sh -nosound -opengl'
RESULT: <xxx>@<xxx>:~/PythonAPI_0.9.10/util$ python3 test_connection.py 
Failed to connect to 127.0.0.1:2000.

gaoyinfeng, 2021-01-09:
@arvindvk555 , hi, I have the same issue when running a container with 0.9.11 version, but it works fine with 0.9.9 version. Do you solve this problem?

corkyw10, 2021-03-04:
@bernatx Could you follow up on this please?

alexkutsan, 2022-04-30:
I have a similar issue on `carlasim/carla:0.9.13`

On `carlasim/carla:latest` it works well. But **carlasim/carla:latest** is not usable by python API and ROS bridge. It says API
version mismatch : 
```
>> import carla
>>> client = carla.Client('192.168.0.191', 2000)
>>> client.load_world('Town02')
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 0.9.12 
WARNING: Simulator API version  = a1b37f7f 
Segmentation fault (core dumped)
```

And if I install `carla=0.9.13` python API works, but ros bridge says that it needs 0.9.12 version: 
```
[bridge-1] [INFO] [1651327208.122137156] [carla_ros_bridge]: Trying to connect to 192.168.0.191:2000
[bridge-1] [FATAL] [1651327208.124152911] [carla_ros_bridge]: CARLA python module version 0.9.12 required. Found: 0.9.13
[bridge-1] [INFO] [1651327208.124590331] [carla_ros_bridge]: Shutting down...
```
With many other python type Errors
My rosbridge version: https://github.com/carla-simulator/ros-bridge/commit/dac9e729b70a3db9da665c1fdb843e96e7e25d04

Shoukat827, 2022-05-18:
@alexkutsan  did you find any supported version carla-ros-bridge for carla 0.9.13
i am installed carla 0.9.13 but ros bride getting crashed.(i needed to go with ROS1 instead of ROS2)
please any advice?

Thanks in advance

closed 2024-05-10 by MarcelPiNacy-CVC