carla-simulator/carla #3600: Map ingestion to pre-built Crala from RoadRunner fails

niyantkumar, 2020-11-18:
I used RoadRunnerR2020b, CARLA_0.9.10.1 and ue4-docker 4.24.3 versions. 

I have followed the guidelines to Export the scene from RoadRunner(RR) and ingest it into the Carla package using the docker image of the Unreal Engine(UE). Here is the link” https://carla.readthedocs.io/en/latest/tuto_A_add_map/” 

I followed every step as they have mentioned for this method to generate the map and also it was success to built a map into Carla as it was showing into the available maps into Carla. But, when I load the map then it is loading and I can even see the traffic light changing color, but the rest is black. I can’t see buildings and lanes etc.

One of my other colleague  tried the deprecated method, which uses Unreal Engine to ingest the map into Carla. But, for him also is the same result. 

I’m also attaching the exported file from RR, a simple scene including two lanes and a junction at the intersection and traffic lights at the junction. 

Can someone please help here??

TKeutgens, 2020-11-19:
I have the same problem here. I'm trying to import the maps from [https://anonymous.4open.science/r/adfc20c6-fb29-4830-a8df-6ac112f9e66e/](url)

I'm working on Windows. I just unpack the files in the Carla root directory and they show up in the map list. But screen stays black.

will-sloan, 2021-01-04:
You might want to try copying the `BasicParameters` folder from one of the given maps into the map you created. This folder should contain the `BP_Sky` object which is probably missing.

TKeutgens, 2021-01-08:
@will-sloan thank's for the hint. I made the changes and the map can be played in the Unreal Editor. 

I also tried copying the files into a pre-build Carla version (0.9.10). The map is shown as available, but each time I try to load the map, the map for Town 03 gets loaded. Do you have any idea how an export from Unreal Editor for pre-build Carla could look work?

will-sloan, 2021-01-12:
I have not tried exporting, but you can select the map you created in the `Content` navigator at the bottom of Unreal Editor. So you could load the map by clicking on it. Is that what you mean? That way the map is loaded even when the simulator isn't playing.

TKeutgens, 2021-01-27:
In the Unreal Editor the map works fine and I am able to run the map. I was wondering if there is a way to export the map from the Unreal Editor to add it to a pre-build Carla version. That way I could share my created maps with everyone.

kanakab, 2021-03-15:
@TKeutgens  were you able to figure this out?

TKeutgens, 2021-03-16:
@kanakab Unfortunately not

kanakab, 2021-03-16:
@TKeutgens I was able to figure out this yesterday.

Go to Carla->Util->BuildTools->./Package.sh --- This step takes a while and it will generate a file named CARLA_Shipping_version_dirty in the folder Carla->Dist

Run the command ./CarlaUe4.sh from the folder '/home/user_name/carla/Dist/CARLA_Shipping_0.9.10-dirty/LinuxNoEditor'  --You will be able to see a carla town in the pre-build version of Carla (To verify the previous step)

Once you are done with all the editing with your map and completed the steps here:https://carla.readthedocs.io/en/latest/tuto_A_add_map/#map-ingestion-in-a-build-from-source
Go to Carla->make package ARGS="--packages=map_name"
This step will also generate one more file in the dist folder by name "map_name_carla_version_number-dirty.tar.gz"

Now copy this .tar.gz file to /home/user_name/carla/Dist/CARLA_Shipping_0.9.10-dirty/LinuxNoEditor/Import/

Run the command ./ImportAssets.sh from the folder '/home/user_name/carla/Dist/CARLA_Shipping_0.9.10-dirty/LinuxNoEditor' -- This will import your map to pre-built version in Carla

To load your custom map:
Run the command ./CarlaUe4.sh from the folder '/home/user_name/carla/Dist/CARLA_Shipping_0.9.10-dirty/LinuxNoEditor'
New Terminal from  '/home/user_name/carla/Dist/CARLA_Shipping_0.9.10-dirty/LinuxNoEditor/PythonAPI/util' and run python config.py -m map_name

If everything is using you can use the .tar.gz file with others for distribution. Let me know if this works for you.

closed 2021-08-05 by corkyw10