carla-simulator/carla #3605: make launch failing on Windows 10

aadimator, 2020-11-19:
Carla Version: 0.9.10
OS: Windows 10
Installed: CMake, make, VS 2017, Windows 10 SDK, Windows 8 SDK, Python

I've successfully built the `PythonAPI` using the command `make PythonAPI`, but when I try to execute the command `make launch`, the following error is shown and I've no idea how to fix this error. Any help would be much appreciated.

```
H:\workspace\NRPU\carla>make launch
-[Setup]: [Batch params]: --boost-toolset msvc-14.1
-[Setup]: Asynchronous jobs:  8
-[Setup]: Boost toolset:      msvc-14.1
-[Setup]: Install directory:  "H:\workspace\NRPU\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "H:\workspace\NRPU\carla\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "H:\workspace\NRPU\carla\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "H:\workspace\NRPU\carla\Build\" --zlib-install-dir "H:\workspace\NRPU\carla\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "H:\workspace\NRPU\carla\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "H:\workspace\NRPU\carla\Build\"
    -[install_rpclib]: A rpclib installation already exists.
    -[install_rpclib]: Delete "H:\workspace\NRPU\carla\Build\rpclib-install\" if you want to force a rebuild.
    -[install_rpclib]: Exiting...
-[Setup]: Installing Google Test...
    -[install_gtest]: [Batch params]: --build-dir "H:\workspace\NRPU\carla\Build\"
    -[install_gtest]: A Google Test installation already exists.
    -[install_gtest]: Delete "H:\workspace\NRPU\carla\Build\gtest-install\" if you want to force a rebuild.
    -[install_gtest]: Exiting...
-[Setup]: Installing "Recast & Detour"...
    -[install_recast]: [Batch params]: --build-dir "H:\workspace\NRPU\carla\Build\"
    -[install_recast]: A "Recast & Detour" installation already exists.
    -[install_recast]: Delete "H:\workspace\NRPU\carla\Build\recast-cdce4e-install\" if you want to force a rebuild.
    -[install_recast]: Exiting...
-[Setup]: Installing Boost...
    -[install_boost]: [Batch params]: --build-dir "H:\workspace\NRPU\carla\Build\" --toolset msvc-14.1 --version 1.72.0 -j 8
    -[install_boost]: A Boost installation already exists.
    -[install_boost]: Delete "H:\workspace\NRPU\carla\Build\boost-1.72.0-install\" if you want to force a rebuild.
    -[install_boost]: Exiting...
        1 file(s) copied.
        1 file(s) copied.
-[Setup]: Installing Xercesc...
    -[install_xercesc]: [Batch params]: --build-dir "H:\workspace\NRPU\carla\Build\"
    -[install_xercesc]: A xerces installation already exists.
    -[install_xercesc]: Delete "H:\workspace\NRPU\carla\Build\xerces-c-3.2.3-install\" if you want to force a rebuild.
    -[install_xercesc]: Exiting...
-[Setup]: Creating "CMakeLists.txt.in"...
-[Setup]:
   ###########
   # SUCCESS #
   ###########

   IMPORTANT!

   All the CARLA library dependences should be installed now.
   (You can remove all "*-src" folders in H:\workspace\NRPU\carla\Build\ directory)

   You only need the ASSET PACK with all the meshes and textures.

   This script provides the assets for CARLA Latest:
   You can download the assets from here:

       http://carla-assets.s3.amazonaws.com/20201103_4b3ab0c.tar.gz

   Unzip it in the "H:\workspace\NRPU\carla\Unreal\CarlaUE4\Content\Carla\" folder.
   If you want another version, search it in H:\workspace\NRPU\carla\Util\ContentVersions.txt.

[... 210 lines cut ...]
  H:/workspace/NRPU/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/InertialMeasurementUnit.cpp(189): warning C4458: declaration of 'Stream' hides class member
  H:\workspace\NRPU\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Sensor/Sensor.h(100): note: see declaration of 'ASensor::Stream'
  H:/workspace/NRPU/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/ObstacleDetectionSensor.cpp(138): warning C4458: declaration of 'Episode' hides class member
  H:\workspace\NRPU\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Sensor/Sensor.h(102): note: see declaration of 'ASensor::Episode'
  H:/workspace/NRPU/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/PixelReader.cpp(96): error C2664: 'bool FPixelReader::WritePixelsToArray(UTextureRenderTarget2D &,TArray<FColor,FDefaultAllocator> &)': cannot convert argument 2 from 'TArray<PixelType,FDefaultAllocator64>' to 'TArray<FColor,FDefaultAllocator> &'
          with
          [
              PixelType=FColor
          ]
  H:/workspace/NRPU/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/SceneCaptureSensor.cpp(340): error C2039: 'AutoExposureCalibrationConstant': is not a member of 'FPostProcessSettings'
  H:\Program Files\Epic Games\UE_4.25\Engine\Source\Runtime\Engine\Classes\Engine/Scene.h(613): note: see declaration of 'FPostProcessSettings'
  H:/workspace/NRPU/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/SceneCaptureSensor.cpp(346): error C2039: 'AutoExposureCalibrationConstant': is not a member of 'FPostProcessSettings'
  H:\Program Files\Epic Games\UE_4.25\Engine\Source\Runtime\Engine\Classes\Engine/Scene.h(613): note: see declaration of 'FPostProcessSettings'
  H:/workspace/NRPU/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/SceneCaptureSensor.cpp(544): error C2039: 'bOverride_AutoExposureCalibrationConstant': is not a member of 'FPostProcessSettings'
  H:\Program Files\Epic Games\UE_4.25\Engine\Source\Runtime\Engine\Classes\Engine/Scene.h(613): note: see declaration of 'FPostProcessSettings'
  H:/workspace/NRPU/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Walker/WalkerController.cpp(28): warning C4458: declaration of 'Character' hides class member
  H:\Program Files\Epic Games\UE_4.25\Engine\Source\Runtime\Engine\Classes\GameFramework/Controller.h(76): note: see declaration of 'AController::Character'
  H:/workspace/NRPU/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Walker/WalkerController.cpp(83): warning C4458: declaration of 'Character' hides class member
  H:\Program Files\Epic Games\UE_4.25\Engine\Source\Runtime\Engine\Classes\GameFramework/Controller.h(76): note: see declaration of 'AController::Character'
  H:/workspace/NRPU/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Walker/WalkerController.cpp(112): warning C4458: declaration of 'Character' hides class member
  H:\Program Files\Epic Games\UE_4.25\Engine\Source\Runtime\Engine\Classes\GameFramework/Controller.h(76): note: see declaration of 'AController::Character'
  H:/workspace/NRPU/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Walker/WalkerController.cpp(126): warning C4458: declaration of 'Character' hides class member
  H:\Program Files\Epic Games\UE_4.25\Engine\Source\Runtime\Engine\Classes\GameFramework/Controller.h(76): note: see declaration of 'AController::Character'
make: *** [CarlaUE4Editor] Error 6
```

BAROBA-3d, 2020-12-10:
I would suggest to git clone again..
There is very little information on building this on Windows.

corkyw10, 2021-03-04:
@aadimator thanks for your patience, are you still facing this issue? There is an updated guide on building CARLA in windows [here](https://carla.readthedocs.io/en/latest/build_windows/)

closed 2021-08-05 by corkyw10