carla-simulator/carla #3606: RuntimeError when triyng to spawn vehicles

salvatoresan, 2020-11-19:
Hi guys, I'm having issues with the spwan_npc script. I run several tests and only a few times I was able to spwan 5 vehicles without any problems. In general, I've got the following error:

[image]

I have no issue  running the manual control script. Besides, I have been able to increase the number of spawned vehicles by increasing the timeout value in the spawn_npc.py script but this workaround does not always work.
Any suggestion?
I'm currently using Carla 0.9.10 built from source on ubuntu

ChicagoPARK, 2021-01-10:
Did you solve it..? I still struggle with it for a long..

salvatoresan, 2021-01-11:
Hi @ChicagoPARK , my workaround to solve this issue was to increse the timeout value in the spawn_npc.py script. 
`104        client.set_timeout(10.0)`
Try to change 10.0 (the default value) in 100.0 and then give it a try by spawning 20 or 30 vehicles. This worked for me but I've avoided to spawn pedestrians.

ChicagoPARK, 2021-01-11:
Well, thank you for sharing your solution. In my case put the button for "play" is the solution. Thank you there

closed 2021-03-04 by corkyw10