carla-simulator/carla #3607: `world.get_map()` after `matplotlib.pyplot.figure()` raises RuntimeError

chauvinSimon, 2020-11-19:
I am repeating multiple times an experiments in a loop.

**Each iteration accesses the current map with `world.get_map()` and saves a path using `matplotlib.pyplot`.**

**Issue: calling `world.get_map()` again after `matplotlib.pyplot.figure()` raises a `RuntimeError`:**

```
Traceback (most recent call last):
  File "/home/me/draft.py", line xxx, in main
    _ = client.get_world().get_map()
RuntimeError: s <= road->GetLength()
```

Importing `matplolib` is not an issue. The problem happens when using `plt.figure()`.

**For you to reproduce, I have written this script**.
- A friend of mine can run it without issue, although we both use `ubuntu20` and use the same version of `matplotlib`.
- Another gets `info != nullptr` as `RuntimeError`.

```
# draft.py
import carla

def main(client: carla.Client, town: str = "Town03", use_plt: bool = False) -> None:
    print("main with town={} and use_plt={}".format(town, use_plt))

    # without the call to get_map(), it works !!
    _ = client.get_world().get_map()

    if use_plt:
        import matplotlib.pyplot as plt
        # error even when turning the interactive mode off with `plt.ioff()`
        plt.figure()

if __name__ == '__main__':
    c = carla.Client('127.0.0.1', 2000)
    main(client=c)  # ok
    main(client=c, use_plt=True)  # ok
    main(client=c)  # fails whatever use_plt
```

---

Additional notes about my system:

```
*************  create fresh conda env ****************
conda create -n myenv python=3.7
yes
conda activate myenv
cd carla09100/PythonAPI/carla/dist/
python -m easy_install carla-0.9.10-py3.7-linux-x86_64.egg
conda install matplotlib
yes
python draft.py
```

```
*************  check carla installation downloaded from website ****************
python -c "import carla; c=carla.Client('127.0.0.1', 2000); print(c.get_server_version()); print(c.get_client_version())"
->
0.9.10
0.9.10
```

```
*************  check installed packages ****************
conda list

# packages in environment at /home/my_name/anaconda3/envs/myenv:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                        main
blas                      1.0                         mkl
ca-certificates           2020.10.14                    0
carla                     0.9.10                   pypi_0    pypi
certifi                   2020.6.20          pyhd3eb1b0_3
cycler                    0.10.0                   py37_0
dbus                      1.13.18              hb2f20db_0
expat                     2.2.10               he6710b0_2
fontconfig                2.13.0               h9420a91_0
freetype                  2.10.4               h5ab3b9f_0
glib                      2.66.1               h92f7085_0
gst-plugins-base          1.14.0               hbbd80ab_1
gstreamer                 1.14.0               hb31296c_0
icu                       58.2                 he6710b0_3
intel-openmp              2020.2                      254
jpeg                      9b                   h024ee3a_2
kiwisolver                1.3.0            py37h2531618_0
lcms2                     2.11                 h396b838_0
ld_impl_linux-64          2.33.1               h53a641e_7
libedit                   3.1.20191231         h14c3975_1
libffi                    3.3                  he6710b0_2
libgcc-ng                 9.1.0                hdf63c60_0
libpng                    1.6.37               hbc83047_0
libstdcxx-ng              9.1.0                hdf63c60_0
libtiff                   4.1.0                h2733197_1
libuuid                   1.0.3                h1bed415_2
libxcb                    1.14                 h7b6447c_0
libxml2                   2.9.10               hb55368b_3
lz4-c                     1.9.2                heb0550a_3
matplotlib                3.3.2                         0
matplotlib-base           3.3.2            py37h817c723_0
mkl                       2020.2                      256
mkl-service               2.3.0            py37he904b0f_0
mkl_fft                   1.2.0            py37h23d657b_0
mkl_random                1.1.1            py37h0573a6f_0
ncurses                   6.2                  he6710b0_1
numpy                     1.19.2           py37h54aff64_0
numpy-base                1.19.2           py37hfa32c7d_0
olefile                   0.46                     py37_0
openssl                   1.1.1h               h7b6447c_0
pcre                      8.44                 he6710b0_0
pillow                    8.0.1            py37he98fc37_0
pip                       20.2.4           py37h06a4308_0
pyparsing                 2.4.7                      py_0
pyqt                      5.9.2            py37h05f1152_2
python                    3.7.9                h7579374_0
python-dateutil           2.8.1                      py_0
qt                        5.9.7                h5867ecd_1
readline                  8.0                  h7b6447c_0
setuptools                50.3.1           py37h06a4308_1
sip                       4.19.8           py37hf484d3e_0
six                       1.15.0           py37h06a4308_0
sqlite                    3.33.0               h62c20be_0
tk                        8.6.10               hbc83047_0
tornado                   6.0.4            py37h7b6447c_1
wheel                     0.35.1             pyhd3eb1b0_0
xz                        5.2.5                h7b6447c_0
zlib                      1.2.11               h7b6447c_3
zstd                      1.4.5                h9ceee32_0
```

```
************* check ubuntu distribution ***********
lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.1 LTS
Release:	20.04
Codename:	focal
```

chauvinSimon, 2020-11-19:
Maybe the following details about my project (much longer than the small script posted above) may be relevant too:

- Following the call to `plt.figure()`:
  - `map=get_map()` does make things break
  - In addition `map.name` is correct
  - **But `map.get_waypoint(loc)` generates a waypoint whose `transform.location` is `>50m` far from `loc`**. The same distance was `~0.0m` just before the call to `plt.figure()` :exploding_head: :woozy_face:

- I got a lot of these lines at start:
```
WARNING: Traffic sign 458 overlaps a driving lane. Moving out of the road... 
```

marcgpuig, 2020-11-20:
Hi @chauvinSimon,

The same happened to me a time ago using `matplotlib`.

As a workaround I took all the information that I needed and stored it somewhere, to finally (at the end of the execution) import the `matplotlib` library and make all the plots that I want.

Regarding this:
```
RuntimeError: s <= road->GetLength()
```
Looks like your OpenDrive may have some errors or you have a very long road and somehow this is causing a floating-point error somewhere

Finally, let me recommend you to store the map into a variable at the beginning, something like:
```py
carla_map = client.get_world().get_map()
```
Otherwise, on each iteration, it will be querying the map to the server, retrieving the whole OpenDrive as XML text, and parsing it to construct a new `carla::client::Map`, which is quite heavy.

Cheers!

chauvinSimon, 2020-11-20:
Thank you @marcgpuig for your complete answer

closed 2020-11-20 by chauvinSimon

Eduard6421, 2021-07-23:
This issue is still active;
Could we receive any information regarding why this is happening?
As a "hotfix" i have removed the global import of matplotlib and added matplotlib imported in each function which needed it;

rokm, 2022-06-08:
This problem is not a CARLA problem per-se, but rather an unfortunate combination of the following factors:
- using the CARLA client code on linux
- using Qt in the client code; either using `PySide2`/`PySide6`/`PyQt5`/`PyQt6` directly (i.e., to create your GUI), or indirectly (for example as `matplotlib` backend, which is probably what happened in OP's case)
- having the system use locale that uses comma (`,`) as a decimal separator and period (`.`) for thousands separator (several European locales) instead of the other way around

That's because instantiating the `QtCore.QCoreApplication` (or `QtWidgets.QApplication`) (re)sets the program's locale to use the system locale instead of default (C) locale, as noted [here](https://doc.qt.io/qt-5/qcoreapplication.html#locale-settings):
<quote>
_On Unix/Linux Qt is configured to use the system locale settings by default. This can cause a conflict when using POSIX functions, for instance, when converting between data types such as floats and strings, since the notation may differ between locales. To get around this problem, call the POSIX function setlocale(LC_NUMERIC,"C") right after initializing QApplication, QGuiApplication or QCoreApplication to reset the locale that is used for number formatting to "C"-locale._
</quote>

So after `QCoreApplication` is instantiated, the POSIX functions for parsing floating-point values might, depending on locale, start expecting comma as decimal separator, and incorrectly parse the received values, as illustrated by the following example:

<details><summary>carla_test.py</summary>
<p>

```python
import sys
import locale

from PySide2 import QtCore
import carla

print("LC_NUMERIC before QCoreApplication:", locale.getlocale(locale.LC_NUMERIC))
app = QtCore.QCoreApplication()
print("LC_NUMERIC after QCoreApplication:", locale.getlocale(locale.LC_NUMERIC))

# Uncomment the following line for the fix
#locale.setlocale(locale.LC_NUMERIC, "C")  # Set locale/numeric back to neutral
#print("LC_NUMERIC after fix:", locale.getlocale(locale.LC_NUMERIC))

client = carla.Client("localhost", 2000)
world = client.get_world()
town_map = world.get_map()
print("Map:", town_map.name)
```

</p></details>

In my case, the print statements show
```
LC_NUMERIC before QCoreApplication: (None, None)
LC_NUMERIC after QCoreApplication: ('sl_SI', 'UTF-8')
```
so `QCoreApplication` instantiation switched numeric representation to Slovenian locale, which uses comma as decimal separator. This causes misinterpretation of received values that use period as a decimal separator (which therefore end up being off by a factor of thousand).

The resulting symptoms vary from map to map.

For example, when running `Town10HD_Opt` map, calling `get_map` in client yields the following warnings:

```
WARNING: Traffic sign 963 overlaps a driving lane. Moving out of the road... 
WARNING: Failed to find suitable place for signal. 
WARNING: Traffic sign 961 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 958 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 959 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 949 overlaps a driving lane. Moving out of the road... 
WARNING: Traffic sign 943 overlaps a driving lane. Moving out of the road...
```

(and the received positions of traffic signs are likely off by a factor of thousand).

Whereas when running `Town03_Opt` map, the client fails to retrieve the map altogether, with the following error:

```
Traceback (most recent call last):
  File "carla_test.py", line 18, in <module>
    town_map = world.get_map()
RuntimeError: info != nullptr
```

The solution, as suggested by the Qt documentation, is to reset `LC_NUMERIC` back to `C` after instantiating the `QCoreApplication`. And indeed, uncommenting the corresponding lines in the above example code fixes the problem.

In the OP's case, the locale should probably be reset back to C after the first `plt.figure()` call, which also selects and initializes the plotting backend. Or, the problem could be avoided altogether by forcing use of non-Qt backend via [`matplotlib.use()` function](https://matplotlib.org/stable/api/matplotlib_configuration_api.html#matplotlib.use) (by default, `matplotlib` seems to prefer Qt-based backend if one of Qt bindings package is available in the environment).

Eduard6421, 2022-06-08:
Thank you for your in depth answer!