carla-simulator/carla #3609: Carla Build Error

wmpeets, 2020-11-19:
I also have the same error as  issue 3605 using
    CMake 3.19
    make 3.8.1
    git 2.29
    Python 3.9.0 amd 64
    Unreal 4.25.4

make PythonAPI completed successfully and make launch failed.

The File ImagePixelData.h is using TArray64 for the "Pixels" data field and CARLA PixelReader.cpp is using TArray.

.../carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/PixelReader.cpp(96): error C2664: 'bool FPixelReader::WritePixelsToArray(UTextureRenderTarget2D &,TArray<FColor,FDefaultAllocator> &)': cannot convert argument 2 from 'TArray<PixelType,FDefaultAllocator64>' to 'TArray<FColor,FDefaultAllocator> &'
          with
          [
              PixelType=FColor
          ]

-->

qhaas, 2020-11-25:
`Unreal 4.25.4`
Try giving it a whirl with Unreal Engine 4.24, the version supported by CARLA 0.9.9+ (assuming you are building from one of the newer tagged releases) ...  I've had bad luck using unsupported Unreal Engines and there are Unreal Engine specific datatypes mentioned in that error log.

wmpeets, 2020-11-29:
Reverted to 4.24.3.  Now there are other errors.

Boost 1.72.0-install:
\carla\Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies\include\carla/rpc/Response.h(53): error C2665: 'boost::variant<carla::rpc::Response<uint64_t>::error_type,unsigned __int64>::variant': none of the 3 overloads could convert all the argument types

corkyw10, 2021-03-04:
@wmpeets thank you for your patience, are you still facing this issue?

closed 2023-06-02 by stale[bot]