carla-simulator/carla #3610: GlobalShaderCache-GLSL_150.bin is missing when running with -opengl3

Stephanehk, 2020-11-20:
I am attempting to install Carla 0.9.10.1 on Linux. When running "swr CarlaUE4.sh -opengl3", I get the following error:

"LogShaders: Error: The global shader cache file Carla_Sim/Engine/GlobalShaderCache-GLSL_150.bin is missing

Your application is built to load COOKED content No COOKED content was found this usually means you did not cook content for this build."

Upon inspecting the Engine folder, the file really is missing, however, there is a file named " GlobalShaderCache-GLSL_430.bin". Is there some other version of Carla that needs to be compiled in order to use opengl3? I am not quite sure what is causing this error.

qhaas, 2020-11-24:
I could be wrong, but I think it only supports opengl 4.  CARLA 0.9.9+ is based on UnrealEngine 4.24, which [appears to have dropped opengl 3.x](https://docs.unrealengine.com/en-US/Engine/Rendering/SupportedRenderingFeatures/index.html):
`GL 3.3+ have been removed for in 4.23.`

KunanonxGithub, 2020-11-26:
@qhaas  Hello, could you help me throught this question pls? I am so curious about how physic in CARLA actually work because I want to use CARLA and PID control to develop Adaptive Cruise Control and I think it need information such as vehicle mass (which is known in document) but b, air coefficient drag force, espicially relationship between throttle and velocity of CARLA car. #3625 is unknown

Stephanehk, 2020-11-26:
@qhaas I tried using CARLA 0.9.5 and had the same error. Was OpenGL 3 support removed from all versions?

qhaas, 2020-11-26:
Clarification, I'm not affiliated with CARLA; I'm just a user.

Looks like using [`-opengl`](https://carla.readthedocs.io/en/0.9.10/adv_rendering_options/#vulkan-vs-opengl) will trigger using opengl 4.3 instead of vulkan for newer versions of CARLA.

CARLA 0.9.5 uses [UnrealEngine 4.21](https://github.com/carla-simulator/carla/blob/0.9.5/Unreal/CarlaUE4/CarlaUE4.uproject#L3) and appears to [target only OpenGL 4.3](https://github.com/carla-simulator/carla/blob/0.9.5/Unreal/CarlaUE4/Config/DefaultEngine.ini#L94).  Looking at the [UnrealEngine 4.21 source](https://github.com/EpicGames/UnrealEngine/blob/4.21.2-release/Engine/Config/BaseEngine.ini#L1954), it doesn't seem to list OpenGL 3 as a 'TargetedRHIs'.

corkyw10, 2021-03-04:
@Axel1092 could you follow up on this please?

Axel1092, 2021-03-15:
Hi all, just to clarify things a little, as @qhaas said OpenGL 3 has been deprecated for some time in unreal engine and therefore you won't get it to work with it. As newer versions come up older render API are being deprecated. CARLA 0.9.11 is probably the last version to support OpenGL 4, making vulkan the only render API available for Linux systems as Unreal has deprecated OpenGL entirely in the latest versions.

sxppro, 2022-11-17:
I am current using the latest version (0.9.13), however, when I launch CARLA without any options I encounter a similar error (except the file is called **GlobalShaderCache-GLSL_150_ES31.bin**). In the directory I actually have a file called **GlobalShaderCache-SF_VULKAN_SM5.bin**. Why is CARLA referring to a different global shader cache?

sxppro, 2022-11-17:
> I am current using the latest version (0.9.13), however, when I launch CARLA without any options I encounter a similar error (except the file is called **GlobalShaderCache-GLSL_150_ES31.bin**). In [...]

Never mind I solved my issue - it was because I hadn't installed vulkan-utils

closed 2026-11-13 by carla-triage[bot]