carla-simulator/carla #3611: Failed to make PythonAPI in Windows

Yifanny, 2020-11-20:
Carla version: 0.9.10
OS: Windows 10
Compiler: MSVC 19.16.27044
CMake version: 3.12

Error messages are shown as the following figure. As shown at the bottom of the figure, I have installed python 3.7. The possible causes indicate that the python is not installed. Any help would be much appreciated.
[image]

aadimator, 2020-11-21:
It's trying to execute the `py` command, while in windows, `python` command is used to access Python. I worked around this issue by going to the Python installation directory, making a copy of `python.exe`, and renaming it to `py.exe`, so I am able to run `py` command from the terminal. This solved the issue for me.

[image]

Yifanny, 2020-11-21:
> It's trying to execute the `py` command, while in windows, `python` command is used to access Python. I worked around this issue by going to the Python installation directory, making a copy of [...]

Many thanks for your help. There is still other error. The error is _failed to find XercesC(missing: XercesC_VERSION)_. I have no idea to solve it. Should I install XercesC manually? Have you encounter such a problem? Many thanks in advance.
[image]

corkyw10, 2021-03-04:
@Yifanny thank you for your patience. Are you still experiencing this issue?

closed 2023-06-02 by stale[bot]