carla-simulator/carla #3615: Spawn failed because of invalid actor description

ZhenWang-GoodST, 2020-11-21:
I tried to spawn a car and got "Spawn failed because of invalid actor description'" in the terminal which launches the UEeditor.
I use the command "python manual_control.py --filter cleancar", the "cleancar" was created through MAYA by myself, just following the official turtorials in carla documentation, and my car can be found in the bluprints libraries. I'd like to know is there anything else for to do to spawn a customized car  ?

corkyw10, 2021-03-04:
@ZhenWang-GoodST Thanks for your patience, are you still facing this issue? If so can you tell me what version of carla you are using and the OS?

XGodina, 2021-03-05:
Hi, @ZhenWang-GoodST 
Do you add your car to the BP_vehicle factory. This BP, contains all cars that you after spawning with manual_control.py or spawn_npc.py

closed 2023-06-02 by stale[bot]