carla-simulator/carla #3616: Using mouse to control camera perspective in manual_control.py client

GustavoSilvera, 2020-11-21:
Hello Carla team!

I am working with the python clients for `manual_control.py` and was wondering if there is a good way to move the perspective of the user (or camera) around with the mouse. Specifically a mechanism like the simulator's `spectator` view but on the client pygame instance would be ideal!

I tried something with destroying and reinitializing the camera sensor but this is very slow. Similarly flooding the server with `set()` requests for `set_transform()` is also very slow. I'm not sure of a better way to do this. 

Any help is appreciated!

Axel1092, 2021-03-04:
Hi @GustavoSilvera, you could use pygame to detect position and drag of the mouse to change the camera orientation accordingly  with the `set_transform()` function you mention but there is no better way to do this.

GustavoSilvera, 2021-03-04:
Hi @Axel1092, thanks for your suggestion. Unfortunately I already tried something similar to this and found that it does not suit my purposes well as it is very laggy with a moving vehicle and I was looking into VR support.

In this regard I've transitioned to using the UE4 game itself (simulator 'server') as the main viewpoint since UE4 has native VR support and so far this approach has been successful.

closed 2021-03-04 by GustavoSilvera