carla-simulator/carla #3617: Document page for PythonAPI appears to be  broken.

android18-rr, 2020-11-22:
A part of the document page for PythonAPI (https://carla.readthedocs.io/en/latest/python_api/) is seems to be broken in its layout.
The part before the "carla.CollisionEvent" section is normal, however, the part after that appears to be broken.
I tried to view the page using some browsers, Firefox, Safari, and Chronium but the page appears to be broken in all these browsers. Is there a problem in buiding the documents ?

tinmodeHuang, 2020-11-22:
me, too.
that have been happening since yesterday.

ebadi, 2020-11-23:
Removing `</p><p>` in [doc_gen.py](https://github.com/carla-simulator/carla/blob/dev/PythonAPI/docs/doc_gen.py#L97) resolves the issue. The fix is part of my recent [pull request](https://github.com/carla-simulator/carla/pull/3614).

tinmodeHuang, 2020-11-23:
> Removing `</p><p>` in [doc_gen.py](https://github.com/carla-simulator/carla/blob/dev/PythonAPI/docs/doc_gen.py#L97) resolves the issue. The fix is a part of my recent [pull request](https://github.com/carla-simulator/carla/pull/3614).

I check out the doc through browser,  can the solution above apply to my situation?

ebadi, 2020-11-23:
@tinmodeHuang I think so. the doc_gen.py script build the [Docs/python_api.md](https://github.com/carla-simulator/carla/blob/dev/PythonAPI/docs/doc_gen.py#L729). The markdown file is later used by mkdocs application to generate [the broken page](https://carla.readthedocs.io/en/latest/python_api/) you mentioned.

tinmodeHuang, 2020-11-23:
> @tinmodeHuang I think so. the doc_gen.py script build the [Docs/python_api.md](https://github.com/carla-simulator/carla/blob/dev/PythonAPI/docs/doc_gen.py#L729). The markdown file is later used by [...]

that is to say, the method just help those out with reading the doc in the type .md, there has been no solution to help us who read the doc through browser

ebadi, 2020-11-23:
The doc in the HTML format (readable with browser) is created from the .md file.

tinmodeHuang, 2020-11-23:
> The doc in the HTML format (readable with browser) is created from the .md file.

I mean, the doc refers to offical documentation, I'm sorry for making it clear.
```bash
(python37) E:\安装包\carla-dev\carla-dev\PythonAPI\docs>python doc_gen.py
Generating PythonAPI documentation...
Generating PythonAPI snipets...
Done snipets!
Traceback (most recent call last):
  File "doc_gen.py", line 735, in <module>
    main()
  File "doc_gen.py", line 730, in main
    md_file.write(docs.gen_markdown())
  File "doc_gen.py", line 720, in gen_markdown
    return join([self.gen_body()], '\n').strip()
  File "doc_gen.py", line 714, in gen_body
    append_code_snipets(md)
  File "doc_gen.py", line 270, in append_code_snipets
    os.remove(snipets_path)
PermissionError: [WinError 32] 另一个程序正在使用此文件，进程无法访问。: 'E:\\安装包\\carla-dev\\carla-dev\\PythonAPI\\docs\\../../Docs/python_api_snipets.md'
```

marcgpuig, 2020-11-25:
Hi, @android18-rr @tinmodeHuang
This has been a quick-fixed in `master` (5bd3dab1) until we test @ebadi's pull request.
The docs should work fine now, especially the [Python API reference](https://carla.readthedocs.io/en/latest/python_api/).

Closing for now, thanks :)

closed 2020-11-25 by marcgpuig

tinmodeHuang, 2020-11-25:
> Hi, @android18-rr @tinmodeHuang
> This has been a quick-fixed in `master` ([5bd3dab](https://github.com/carla-simulator/carla/commit/5bd3dab1013df554c0198662e0ceb50b7857feba)) until we test @ebadi's pull request.
> The docs should work fine now, especially the [Python API reference](https://carla.readthedocs.io/en/latest/python_api/).
> 
> Closing for now, thanks :)

as you said, it actually works again