carla-simulator/carla #3618: Carla fails with large number of iterations

yasser-h-khalil, 2020-11-22:
I ran spawn_npc.py for a large number of iterations using the nighty version (above 0.9.10.1). On the client side I got:

`RuntimeError: time-out of 10000ms while waiting for the simulator, make sure the simulator is ready and connected to 127.0.0.1:2000`

and on the simulator side I got: 

`terminating with uncaught exception of type clmdep_msgpack::v1::type_error: std::bad_cast
terminating with uncaught exception of type clmdep_msgpack::v1::type_error: std::bad_cast
Signal 6 caught.
Signal 6 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
terminating with uncaught exception of type clmdep_msgpack::v1::type_error: std::bad_cast
Signal 6 caught.
CommonUnixCrashHandler: Signal=6
terminating with uncaught exception of type clmdep_msgpack::v1::type_error: std::bad_cast
terminating with uncaught exception of type clmdep_msgpack::v1::type_error: std::bad_castterminating with uncaught exception of type clmdep_msgpack::v1::type_error: std::bad_castSignal 6 caught.
terminating with uncaught exception of type clmdep_msgpack::v1::type_error: std::bad_cast

Signal 6 caught.
Signal 6 caught.
Signal 6 caught.
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=123824 LargeMemoryPoolOffset=254960 
Engine crash handling finished; re-raising signal 6 for the default handler. Good bye.
Aborted (core dumped)
`

Attaching the code:

`    vehicles_list = []
    walkers_list = []
    all_id = []

    client = carla.Client(args.host, args.port)
    client.set_timeout(10.0)
    synchronous_master = False
    random.seed(args.seed if args.seed is not None else int(time.time()))
    synchronization = True

    try:
        world = client.get_world()

        traffic_manager = client.get_trafficmanager(args.tm_port)
        traffic_manager.set_global_distance_to_leading_vehicle(1.0)
        if args.hybrid:
            traffic_manager.set_hybrid_physics_mode(True)
        if args.seed is not None:
            traffic_manager.set_random_device_seed(args.seed)

        if synchronization:
            settings = world.get_settings()
            traffic_manager.set_synchronous_mode(True)
            if not settings.synchronous_mode:
                print("Entered Here")
                synchronous_master = True
                settings.synchronous_mode = True
                settings.fixed_delta_seconds = 0.05
                world.apply_settings(settings)
            else:
                synchronous_master = False

        blueprints = world.get_blueprint_library().filter(args.filterv)
        blueprintsWalkers = world.get_blueprint_library().filter(args.filterw)

        if args.safe:
            blueprints = [x for x in blueprints if int(x.get_attribute('number_of_wheels')) == 4]
            blueprints = [x for x in blueprints if not x.id.endswith('isetta')]
            blueprints = [x for x in blueprints if not x.id.endswith('carlacola')]
            blueprints = [x for x in blueprints if not x.id.endswith('cybertruck')]
            blueprints = [x for x in blueprints if not x.id.endswith('t2')]
[... 158 lines cut ...]

            time.sleep(1)

            if synchronization:
                settings = world.get_settings()
                settings.synchronous_mode = False
                settings.fixed_delta_seconds = None
                world.apply_settings(settings)

            print('\ndestroying %d vehicles' % len(vehicles_list))
            client.apply_batch([carla.command.DestroyActor(x) for x in vehicles_list])

            # stop walker controllers (list is [controller, actor, controller, actor ...])
            for i in range(0, len(all_id), 2):
                all_actors[i].stop()

            print('\ndestroying %d walkers' % len(walkers_list))
            client.apply_batch([carla.command.DestroyActor(x) for x in all_id])

            time.sleep(5)`

Command used to run the simulator is ` DISPLAY= ./CarlaUE4.sh -opengl -carla-rpc-port=2000 -carla-streaming-port=0`

This is giving me a lot of pressure. I have been stuck on this for weeks now. Help is much appreciated! 
Thanks in advance.

KunanonxGithub, 2020-11-26:
@bernatx Hello, are you developer of CARLA? could you or your colleague help me throught this question pls? I am so curious about how physic in CARLA actually work because I want to use CARLA and PID control to develop Adaptive Cruise Control and I think it need information such as vehicle mass (which is known in doc) and b,which is air coefficient drag force, espicially relationship formula between throttle and velocity of CARLA car. #3625

KangChou, 2020-12-11:
actorlist = world.spawn_actor(bp, spoint).set_autopilot(enabled=True)
RuntimeError: Spawn failed because of collision at spawn position

corkyw10, 2021-03-04:
@bernatx is there an update on this issue?

coutyou, 2022-02-26:
I met this problem, too :(

closed 2023-06-02 by stale[bot]