carla-simulator/carla #3621: make PythonAPI  fails for Windows 10,

frodozet, 2020-11-23:
linker fails building libcarla.obj in make PythonAPI
what could be the reason?

where to show for those unresolved external symbols???

Platform: Windows 10
VS 2017
python 3.7.9 64bit
carla: 0.9.10.1-122-g064760da

make launch OK!
make PythonAPI ERROR!
unresolved LNK2001

TrafficManager.cpp
dependencies/include\carla/trafficmanager/RandomGenerator.h(14): warning C4244: 'argument': conversion from 'const uint64_t' to 'unsigned int', possible loss of data
dependencies/include/carla\trafficmanager\TrafficManager.cpp(72): warning C4244: 'initializing': conversion from 'SOCKET' to 'int', possible loss of data
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Idependencies/include "-IC:\Program Files\Python37\include" "-IC:\Program Files\Python37\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\cppwinrt" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\cppwinrt" /EHsc /Tpdependencies/include/carla\trafficmanager\TrafficManagerLocal.cpp /Fobuild\temp.win-amd64-3.7\Release\dependencies/include/carla\trafficmanager\TrafficManagerLocal.obj /experimental:external /external:I dependencies/include/system /DBOOST_ALL_NO_LIB /DBOOST_PYTHON_STATIC_LIB /DBOOST_ERROR_CODE_HEADER_ONLY /D_WIN32_WINNT=0x0600 /DHAVE_SNPRINTF /DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true
TrafficManagerLocal.cpp
dependencies/include\carla/trafficmanager/RandomGenerator.h(14): warning C4244: 'argument': conversion from 'const uint64_t' to 'unsigned int', possible loss of data
dependencies/include/carla\trafficmanager\TrafficManagerLocal.cpp(157): warning C4267: 'initializing': conversion from 'size_t' to 'unsigned long', possible loss of data
dependencies/include/carla\trafficmanager\TrafficManagerLocal.cpp(164): warning C4267: '=': conversion from 'size_t' to 'unsigned long', possible loss of data
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Idependencies/include "-IC:\Program Files\Python37\include" "-IC:\Program Files\Python37\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\include" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\cppwinrt" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\cppwinrt" /EHsc /Tpdependencies/include/carla\trafficmanager\TrafficManagerRemote.cpp /Fobuild\temp.win-amd64-3.7\Release\dependencies/include/carla\trafficmanager\TrafficManagerRemote.obj /experimental:external /external:I dependencies/include/system /DBOOST_ALL_NO_LIB /DBOOST_PYTHON_STATIC_LIB /DBOOST_ERROR_CODE_HEADER_ONLY /D_WIN32_WINNT=0x0600 /DHAVE_SNPRINTF /DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true
TrafficManagerRemote.cpp
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX86\x64\link.exe /nologo /INCREMENTAL:NO /LTCG /DLL /MANIFEST:EMBED,ID=2 /MANIFESTUAC:NO /LIBPATH:dependencies/lib "/LIBPATH:C:\Program Files\Python37\libs" "/LIBPATH:C:\Program Files\Python37\PCbuild\amd64" "/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\ATLMFC\lib\x64" "/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\lib\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.18362.0\ucrt\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.18362.0\um\x64" "/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\ATLMFC\lib\x64" "/LIBPATH:C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\lib\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\lib\um\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.18362.0\ucrt\x64" "/LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.18362.0\um\x64" /EXPORT:PyInit_libcarla build\temp.win-amd64-3.7\Release\source/libcarla/libcarla.obj build\temp.win-amd64-3.7\Release\dependencies/include/carla\Buffer.obj build\temp.win-amd64-3.7\Release\dependencies/include/carla\Exception.obj build\temp.win-amd64-3.7\Release\dependencies/include/carla\FileSystem.obj build\temp.win-amd64-3.7\Release\dependencies/include/carla\StringUtil.obj build\temp.win-amd64-3.7\Release\dependencies/include/carla\client\Actor.obj build\temp.win-amd64-3.7\Release\dependencies/include/carla\client\ActorAttribute.obj build\temp.win-amd64-3.7\Release\dependencies/include/carla\client\ActorBlueprint.obj build\temp.win-amd64-3.7\Release\dependencies/include/carla\client\ActorList.obj build\temp.win-amd64-3.7\Release\dependencies/include/carla\client\BlueprintLibrary.obj build\temp.win-amd64-3.7\Release\dependencies/include/carla\client\DebugHelper.obj build\temp.win-amd64-3.7\Release\dependencies/include/carla\client\Junction.obj build\temp.win-amd64-3.7\Release\dependencies/include/carla\client\LaneInvasionSensor.obj build\temp.win-amd64-3.7\Release\dependencies/include/carla\client\Light.obj build\temp.win-amd64-3.7\Release\dependencies/include/carla\client\LightManager.obj build\temp.win-amd64-3.7\Release\dependencies/include/carla\client\Map.obj build\temp.win-amd64-3.7\Release\dependencies/include/carla\client\ServerSideSensor.obj build\temp.win-amd64-3.7\Release\dependencies/include/carla\client\TimeoutException.obj build\temp.win-amd64-3.7\Release\dependencies/include/carla\client\TrafficLight.obj build\temp.win-amd64-3.7\Release\dependencies/include/carla\client\TrafficSign.obj build\temp.win-amd64-3.7\Release\dependencies/include/carla\client\Vehicle.obj build\temp.win-amd64-3.7\Release\dependencies/include/carla\client\Walker.obj build\temp.win-amd64-3.7\Release\dependencies/include/carla\client\WalkerAIController.obj build\temp.win-amd64-3.7\Release\dependencies/include/carla\client\Waypoint.obj build\temp.win-amd64-3.7\Release\dependencies/include/carla\client\World.obj build\temp.win-amd64-3.7\Release\dependencies/include/carla\client\detail\ActorFactory.obj build\temp.win-amd64-3.7\Release\dependencies/include/carla\client\detail\ActorState.obj build\temp.win-amd64-3.7\Release\dependencies/include/carla\client\detail\ActorVariant.obj build\temp.win-amd64-3.7\Release\dependencies/include/carla\client\detail\Client.obj build\temp.win-amd64-3.7\Release\dependencies/include/carla\client\detail\Episode.obj build\temp.win-amd64-3.7\Release\dependencies/ [... cut ...]
osm2odr.lib(SUMOSAXReader.obj) : error LNK2001: unresolved external symbol inflateInit2_
osm2odr.lib(OutputDevice_File.obj) : error LNK2001: unresolved external symbol inflateInit2_
build\lib.win-amd64-3.7\carla\libcarla.cp37-win_amd64.pyd : fatal error LNK1120: 87 unresolved externals
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\VC\\Tools\\MSVC\\14.16.27023\\bin\\HostX86\\x64\\link.exe' failed with exit status 1120

-[BuildPythonAPI]: Carla lib for python has been successfully installed in

bernatx, 2020-11-30:
Hi,
Usually when we see this error about _unresolved external symbol_ on _boost::python_, it is caused by a wrong version of Python or a 32 bits version instead of the 64 bits version of Python required. Can you check that?

frodozet, 2020-11-30:
Yes,
Python 3.7.9 and 3.7.2 checked and both failed. Is there a lib dll or special file in the python package i can or should look for?

frodozet, 2020-12-01:
Hi
no 32bit version

Eikeike, 2020-12-02:
I've had the exact same issue. I don't know if it will help in your case, but here's what I did:
 
1. Delete Build and PythonAPI/carla/build - Directories
2. Downgrade Python from 3.7.9 to **3.7.7** (it's easiest to completely uninstall python and install the version you need)
3. make clean
4. make rebuild

frodozet, 2020-12-06:
doesn't change anything.
Same errors.
Does anyone  have an idea about these errors? 
They probably are just correct and a hint ?
" void __cdecl boost::python::converter::registry::insert(struct _object * ( ..."
Where ist this symbol defined?

Manue26, 2021-02-24:
Hi, 

I have exactly the same issue. did you resolve your problem ? 

I'm using 
Platform: Windows 10
VS 2017 (I also have VS 2019 for other projects)
python 3.9 64bit (but I also have python 3.7 on my computer) 
carla: 0.9.10.1

bernatx, 2021-02-24:
Hi,

Another cause of that problem is that when you build the first time CARLA, when it setup the boost files it takes a python version installed (if you have several it can take one that you don't want), and then later when compiling the PythonAPI the linker is using another version of Python, so it can not link different versions. 

So, be sure that by default the same version of Python is found always, and it is 64 bits.

You can check the python that boost was using looking the files here:
CARLA\Build\boost-1.72.0-install\lib\libboost_python**37**-vc141-mt-x64-1_72.lib 
the ..37.. is the version of Python, in this case 3.7

To rebuild, just remove the **CARLA\Build\boost-1.72.0-install** folder and try to build PythonAPI again with **make PythonAPI**.

xyxu1024, 2021-04-02:
same issue!
```shell
libcarla.obj : error LNK2001: 无法解析的外部符号 "void __cdecl boost::python::throw_error_already_set(void)" (?throw_error_already_set@python@boost@@YAXXZ)
libcarla.obj : error LNK2001: 无法解析的外部符号 "struct boost::python::self_ns::self_t boost::python::self_ns::self" (?self@self_ns@python@boost@@3Uself_t@123@A)
libcarla.obj : error LNK2001: 无法解析的外部符号 "void * __cdecl boost::python::converter::get_lvalue_from_python(struct _object *,struct boost::python::converter::registration const &)" (?get_lvalue_from_python@converter@python@boost@@YAPEAXPEAU_object@@AEBUregistration@123@@Z)
libcarla.obj : error LNK2001: 无法解析的外部符号 "struct boost::python::converter::registration const & __cdecl boost::python::converter::registry::lookup_shared_ptr(struct boost::python::type_info)" (?lookup_shared_ptr@registry@converter@python@boost@@YAAEBUregistration@234@Utype_info@34@@Z)
libcarla.obj : error LNK2001: 无法解析的外部符号 "struct boost::python::converter::registration const * __cdecl boost::python::converter::registry::query(struct boost::python::type_info)" (?query@registry@converter@python@boost@@YAPEBUregistration@234@Utype_info@34@@Z)
libcarla.obj : error LNK2001: 无法解析的外部符号 "void __cdecl boost::python::converter::registry::push_back(void * (__cdecl*)(struct _object *),void (__cdecl*)(struct _object *,struct boost::python::converter::rvalue_from_python_stage1_data *),struct boost::python::type_info,struct _typeobject const * (__cdecl*)(void))" (?push_back@registry@converter@python@boost@@YAXP6APEAXPEAU_object@@@ZP6AX0PEAUrvalue_from_python_stage1_data@234@@ZUtype_info@34@P6APEBU_typeobject@@XZ@Z)
libcarla.obj : error LNK2001: 无法解析的外部符号 "void __cdecl boost::python::converter::registry::insert(struct _object * (__cdecl*)(void const *),struct boost::python::type_info,struct _typeobject const * (__cdecl*)(void))" (?insert@registry@converter@python@boost@@YAXP6APEAU_object@@PEBX@ZUtype_info@34@P6APEBU_typeobject@@XZ@Z)
libcarla.obj : error LNK2001: 无法解析的外部符号 "void __cdecl boost::python::converter::registry::insert(void * (__cdecl*)(struct _object *),void (__cdecl*)(struct _object *,struct boost::python::converter::rvalue_from_python_stage1_data *),struct boost::python::type_info,struct _typeobject const * (__cdecl*)(void))" (?insert@registry@converter@python@boost@@YAXP6APEAXPEAU_object@@@ZP6AX0PEAUrvalue_from_python_stage1_data@234@@ZUtype_info@34@P6APEBU_typeobject@@XZ@Z)
libcarla.obj : error LNK2001: 无法解析的外部符号 "struct boost::python::converter::registration const & __cdecl boost::python::converter::registry::lookup(struct boost::python::type_info)" (?lookup@registry@converter@python@boost@@YAAEBUregistration@234@Utype_info@34@@Z)
libcarla.obj : error LNK2001: 无法解析的外部符号 "public: void __cdecl boost::python::converter::shared_ptr_deleter::operator()(void const *)" (??Rshared_ptr_deleter@converter@python@boost@@QEAAXPEBX@Z)
libcarla.obj : error LNK2001: 无法解析的外部符号 "public: __cdecl boost::python::converter::shared_ptr_deleter::~shared_ptr_deleter(void)" (??1shared_ptr_deleter@converter@python@boost@@QEAA@XZ)
libcarla.obj : error LNK2001: 无法解析的外部符号 "public: __cdecl boost::python::converter::shared_ptr_deleter::shared_ptr_deleter(class boost::python::handle<struct _object>)" (??0shared_ptr_deleter@converter@python@boost@@QEAA@V?$handle@U_object@@@23@@Z)
libcarla.obj : error LNK2001: 无法解析的外部符号 "void __cdecl boost::python::converter::void_result_from_python(struct _object *)" (?void_result_from_python@converter@python@boost@@YAXPEAU_object@@@Z)
libcarla.obj : error LNK2001: 无法解析的外部符号 "void * __cdecl boost::python::converter::rvalue_from_python_stage2(struct _object *,struct boost::python::converter::rvalue_from_python_stage1_data &,struct boost::python::converter::registration const &)" (?rvalue_from_python_stage2@converter@python@boost@@YAPEAXPEAU_object@@AEAUrvalue_from_python_stage1_data@123@AEBUregistration@123@@Z)
libcarla.obj : error LNK2001: 无法解析的外部符号 "struct boost::python::converter::rvalue_from_python_stage1_data __cdecl boost::python::converter::rvalue_from_python_stage1(struct _object *,struct boost::python::converter::registration const &)" (?rvalue_from_python_stage1@converter@python@boost@@YA?AUrvalue_from_python_stage1_data@123@PEAU_object@@AEBUregistration@123@@Z)
libcarla.obj : error LNK2001: 无法解析的外部符号 "public: __cdecl boost::python::converter::detail::arg_to_python_base::arg_to_python_base(void const volatile *,struct boost::python::converter::registration const &)" (??0arg_to_python_base@detail@converter@python@boost@@QEAA@PEDXAEBUregistration@234@@Z)
libcarla.obj : error LNK2001: 无法解析的外部符号 "struct _object * __cdecl boost::python::converter::do_return_to_python(char const *)" (?do_return_to_python@converter@python@boost@@YAPEAU_object@@PEBD@Z)
libcarla.obj : error LNK2001: 无法解析的外部符号 "struct _object * __cdecl boost::python::converter::do_return_to_python(struct _object *)" (?do_return_to_python@converter@python@boost@@YAPEAU_object@@PEAU4@@Z)
libcarla.obj : error LNK2001: 无法解析的外部符号 "bool __cdecl boost::python::converter::implicit_rvalue_convertible_from_python(struct _object *,struct boost::python::converter::registration const &)" (?implicit_rvalue_convertible_from_python@converter@python@boost@@YA_NPEAU_object@@AEBUregistration@123@@Z)
libcarla.obj : error LNK2001: 无法解析的外部符号 "public: struct _typeobject const * __cdecl boost::python::converter::registration::to_python_target_type(void)const " (?to_python_target_type@registration@converter@python@boost@@QEBAPEBU_typeobject@@XZ)
[... cut ...]
libcarla.obj : error LNK2001: 无法解析的外部符号 "class boost::python::api::object __cdecl boost::python::api::getitem(class boost::python::api::object const &,class boost::python::api::object const &)" (?getitem@api@python@boost@@YA?AVobject@123@AEBV4123@0@Z)
build\lib.win-amd64-3.7\carla\libcarla.cp37-win_amd64.pyd : fatal error LNK1120: 77 个无法解析的外部命令
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\bin\\HostX86\\x64\\link.exe' failed with exit status 1120
```

7788boy, 2021-05-16:
Hi, 
I tried the following steps and it works.

1. Remove all the other versions of Python. (I uninstall Anaconda and reinstall Python 3.7.9 from Python.org.)
2. Add Python to path variable.
3. Delete the folders naming with "-install" in CARLA\Build
[image]
4. make clean
5. make PythonAPI

I hope it will help!

Yebihashu, 2021-09-01:
I solved such problem that happened to me, 
A description is given here  https://github.com/carla-simulator/carla/issues/4060#issuecomment-910363370

nnorbert99, 2021-09-04:
i tried  the python 3.8.10 and basically everything so far, even installing on a new PC with fresh windows on it but the same error happend. Anybody got any idea ?

nnorbert99, 2021-09-05:
Guys i really tried everything, i tried building the 0.9.12 and rebuild the unreal engine, other versions of Visual studio but nothing works. Any ide or hint what couse this problem or who should i ask ?

BedirhanKeskin, 2021-10-17:
@nnorbert99 , Could you solve the problem? I also tried but I am having the same issue

BedirhanKeskin, 2021-10-20:
I could solve the problem.

Firstly I was having b2.exe error during "make PythonAPI" which is mentioned in #4534. I downgraded the python version to 3.8.10 from 3.10.xx. I deleted all Carla and python 3.10.xx files and paths then I started from WIndows Build-Part Two: Build CARLA. That solved the b2.exe problem but it starts giving the error of "failed with exit status 1120". After that, I reinstall Make and Cmake. Also don't forget to install python dependencies (setuptools, wheel) again. 

Now I am having an "Out of video memory trying to allocate a rendering resource" error when I launch Carla. My PC has 8GB RAM, i5-4210, and Nvidia Geforce 840M. I guess I need a more powerful PC. 
Thanks

martino92, 2022-03-29:
same issue without solution