carla-simulator/carla #3623: Recommended Way of Working in Unstructured Environments

goksanisil23, 2020-11-24:
Hi,

I wanted to get the opinion of the developers and the community on using Carla for doing simulations for unstructured environments like caves, mountains, forests, etc. 

I understand that the map creation process and the movement of ego and NPCs are heavily dependent on OpenDrive today. But would it be possible to strap the OpenDrive features away from Carla to control the agents in these unstructured environments? 

Thanks in advance

/Goksan

germanros1987, 2021-03-05:
Hi @goksanisil23 

From the perspective of making an unstructured map, you would have to leverage UE4's editor or external tools (perhaps a combination of both). Then you would need to give up all the NPC controllers (Traffic Manager, etc.), and put together your own navigation logic.

You may think that something like the LocalPlanner could be reused since it only depends on some general waypoints that the user could provide as input. However, I believe this may not be the case. The LocalPlanner has been designed and tested in road environments, so other --harsh-- environments may put the LocalPlanner in serious trouble. 

Again, my recommendation is to create your own control logic for those environments leveraging the control API (steering, throttle, etc.). At a minimum you would need:

- A global planner. In your case, this may be A* or something trivial (start and end waypoints?)
- A local planner. The "meat" of the problem... In unstructured environments, you may need a lot of feedback from the environment to adjust your local planner. 
- A controller. You would need to heavily tune the default PID controllers or even introduce a more sophisticated control strategy.

This is not on our agenda yet, but we welcome contributions and collaborations. If there is enough interest we could collaborate.

goksanisil23, 2021-03-06:
Thank you very much for your insightful feedback @germanros1987