carla-simulator/carla #3627: set_autopilot doesn't work after the first call

Stagakis, 2020-11-25:
OS: Ubuntu 20.04
Carla Version: 0.9.10
Python Version: python 3.8.5

Hello, I am trying to create a script for data generation from scratch. While doing so I noticed some weird behaviour from the autopilot. Specifically, if I have a number of actors and I set to each one of them set_autopilot(True), only the first call actually works, resulting in actor1 moving normally while actor2 and actor3 remain stationary. I have a minimal script to reproduce this problem.

This is also true in the case I use a batch to first initialize N vehicles and then create actor1. Actor1 in this case stays stationary while the N vehicles move normally.

https://pastebin.com/B42QRkmw

jackbart94, 2020-11-25:
Hi,

There are a few things that come to mind. First of all, you are using ``world.tick()`` without having synchronous mode activated. Secondly, if you look at ``spawn_npc.py`` you can see that we tick the world after spawning vehicles. That should probably be added as well, to make sure the world has finished spawning an actor before creating more.

Let me know if this helps!

KunanonxGithub, 2020-11-26:
@jackbart94  Hello, are you developer of CARLA? could you or your colleague help me throught this question pls? I am so curious about how physic in CARLA actually work because I want to use CARLA and PID control to develop Adaptive Cruise Control and I think it need information such as vehicle mass (which is known) and b air coefficient drag force espicially relationship between throttle and velocity of CARLA car. https://github.com/carla-simulator/carla/issues/3625

yasser-h-khalil, 2020-11-27:
I noticed in tutorial.py that even though the actors are set to autopilot, they do not actually move!

jackbart94, 2020-11-27:
> I noticed in tutorial.py that even though the actors are set to autopilot, they do not actually move!

We actually fixed ``tutorial.py`` not so long ago, if you switch to `dev` branch it is indeed working.

yasser-h-khalil, 2020-11-27:
Indeed with Carla0.9.10 tutorial.py used to fail and couldn't spawn all 10 vehicles. In the nighty version this was fixed. However, I am talking about the autopilot mode. Even though the vehicles are spawned successfully they don't actually move. I am aware of the recent change in the file (set_autopilot(true)). 

Another thing is that with long iterations carla crashes, I wish if there is a fix for this or a walkaround.

libing0811, 2021-07-08:
> Let me know if this is still happening with CARLA 0.9.11 or latest dev. Also, we have tried long iterations with CARLA and haven't seen any crash.

This problem is still there in 0.9.11, i resolve this problem by add a time.sleep(1) before npc.set_autopilot(True).

closed 2021-11-18 by jackbart94