carla-simulator/carla #3628: Noise for Semantic LiDAR Sensor Question

tiger-bug, 2020-11-25:
Good morning,

According to Read the Docs for the latest version of CARLA, the [semantic LiDAR sensor](https://carla.readthedocs.io/en/latest/ref_sensors/#semantic-lidar-sensor) does not have a **noise_stddev** blueprint attribute like the LiDAR sensor.  Is this something that can be added?  I'm using the LiDAR scanner to build semantically labeled point clouds and I am using the Transform attribute to accomplish this, however I can instead use the IMU Transform matrix instead (since I can add noise to that sensor).  It would be nice to have noise to the LiDAR sensor as well, although I understand that this is pretty minor.   

For reference, I am using the 0.9.10 version of CARLA on an Ubuntu 18.04 operating system.

Thank you!

tiger-bug, 2020-12-18:
I just wanted to update this.  I am converting the **x,y,z** coordinates to spherical coordinates and adding noise to the range.  So 

x = psin(phi)cos(theta)
y = psin(phi)sin(theta)
z=pcos(phi)

I add noise to p 
```python
noise=np.random.normal(loc=0,scale=<noise_level>, size=<number-of-points>)
p+=noise
```
Then convert back to **x,y,z**.

NurbekHalikulov, 2021-02-02:
Hey, 

could you please give more information regarding the noise you added? About the noise_level and number_of_points?

Thanks

DSantosO, 2021-02-02:
Hello @tiger-bug,
The SemanticLidar is thought to be a raw raycasting sensor from which the user can build all his/her needs. This is the reason why it does not incorporate noise nor intensity. If there is something very useful we will think to add it.    
In your case, your approach of adding the noise directly in the client is a good option.

tiger-bug, 2021-02-02:
Good morning @DSantosO ,

That makes sense.  I should have probably closed this earlier.  The noise method shown above seems to be sufficient for my purposes.  Thanks again!

Good morning @NurbekHalikulov ,
  Number of points is depends on the size of the array.  so for a point cloud, **p**, the number of points would be 
```python
p = np.array(lidar_data) # This is the point cloud derived from the scanner.  If you look at /carla-simulator/PythonAPI/examples there should be some good example code for extracting LIDAR 
number_of_points = p.shape[0] #p.shape[0] meaning the number of points in the array, where array shape is (<number of points>, <number of attributes>).  
```
The noise level is up to you.

closed 2021-02-02 by tiger-bug

aczy118, 2021-08-25:
Hi, @tiger-bug 

Were you able to do the spherical projection to 2D plane? If so can you please share your code also? I am also trying to do point cloud segmentation using semantic lidar and trying to modify open3d_lidar.py script, how can I do it for a specific ObjTag like vehicle or building?

DavideCoppola97, 2021-10-05:
> I just wanted to update this. I am converting the **x,y,z** coordinates to spherical coordinates and adding noise to the range. So
>
> x = psin(phi)cos(theta) y = psin(phi)sin(theta) z=pcos(phi)
>
> I add [...]

Hi, I also find myself wanting to add some noise to the semantic lidar, but I don't understand where to get the phi and theta angles, the only thing that the semantic lidar returns is the cosine of the incident angle between the ray, and the normal of the hit object.