carla-simulator/carla #3634: Camera underexposed under Windows, overexposed in Linux in v0.9.10.1

charlie-wartnaby, 2020-11-26:
Building v0.9.10.1 from source under Windows 10, I found that running `carla/PythonAPI/examples/manual.control.py` gave a virtually black screen in the client area that should be showing the CARLA world.

In fact some pixels had non-zero RGB values, I eventually realised, having fruitlessly messed about with pygame versions initially. That led me to #3443 (closed), where it was explained that the camera sensor settings have changed since the last release, and may need manual configuration now.

But in Linux (Ubuntu 18.04), `manual_control.py` apparently worked OK -- well actually, the image was too bright, when I looked again.

I can make things OK in Windows with this kludge to manual_control.py:

```
        self.sensors = [
            ['sensor.camera.rgb', cc.Raw, 'Camera RGB', {'exposure_mode': 'manual', 'shutter_speed': '0.4'}], # CW IDIADA workaround for broken auto-exposure in v0.9.10.1 
```

But if I use that under Linux, the view is overexposed to the point of being almost all white. Not surprising really as a shutter speed of 1/0.4 = 2.5 sec is ridiculously long. In Linux, a `shutter_speed` setting there of 3000 (i.e. 1/3000 sec) gives a reasonable match with the Unreal viewport rendering. Which if anything seems suspiciously fast.

So there seems to be a big mismatch between what happens under Windows, and what happens under Linux. And the default `histogram` exposure mode gives a disastrously dark image under Windows, and a rather overexposed one under Linux.

Many thanks for any suggestions. I'd like my scripts to work under both environments.

Charlie Wartnaby, Applus IDIADA

charlie-wartnaby, 2020-11-26:
So as a workaround for now I've made it fit Linux and Windows with this bodge in `manual_control.py`:

```
        # CW IDIADA kludge: camera exposure was underexposed (virtually black) by default under Windows, and
        # somewhat overexposed under Linux. Setting manual exposure as a workaround to v0.9.10.1 for now.
        camera_reciprocal_shutter_speed = '0.4' if os.name == 'nt' else '3000'
        camera_exposure_mode = 'manual'
        self.sensors = [
            ['sensor.camera.rgb', cc.Raw, 'Camera RGB', {'exposure_mode': camera_exposure_mode, 'shutter_speed': camera_reciprocal_shutter_speed}], # CW IDIADA see above
```

corkyw10, 2021-03-05:
@DSantosO are you able to have a look at this please?

closed 2023-06-02 by stale[bot]