carla-simulator/carla #3635: Custom prop is missing texture

lukase, 2020-11-26:
CARLA version: 0.9.10.1
Platform: Windows 10, Docker for Windows with WSL2 backend

Steps taken:
1. Create custom prop geometry in Blender 2.90
2. Export prop as FBX file, embed texture media in FBX file and apply face smoothing
3. Create import data according to https://carla.readthedocs.io/en/latest/tuto_A_add_props/
4. Run package generation using docker_tools.py
5. Extract generated package into the release directory (WindowsNoEditor)
6. Spawn the prop in a level

Expected behaviour:
Prop shows up with the correct diffuse texture applied

Observed behaviour:
Prop shows up with the default Unreal checkerboard texture

Further information:
- The FBX file renders fine in Windows' Paint 3D
- When using the FBX importer in Unreal Editor in a sample project, the asset is imported correctly (one mesh, one texture, one material) and renders fine
- In both tests above, the original texture file was renamed to make sure it is not loaded via absolute paths
- The asset package generated by docker_tools.py also contains three different assets which probably correspond to mesh, texture and material

What am I missing here?

corkyw10, 2021-03-05:
@XGodina is there any update on this?

mkah91, 2021-04-08:
I am having basically the same issue with CARLA version 0.9.11. @lukase Do you have any updates on this?

mkah91, 2021-04-16:
@lukase Could you share your Blender export settings with me? 
I'm currently struggeling to get the "three different assets which probably correspond to mesh, texture and material" from docker_tools.py. The process finishes with "Package.sh: Success!" but no assets are built.

daohu527, 2024-11-08:
Still have the some problem for now, carla 0.9.13