carla-simulator/carla #3638: no road lines in semantic cloud for carla 0.9.10.1

SimonLeeGit, 2020-11-27:
hi, 
   I'm exciting to use the semantic cloud feature in carla 0.9.10.1. But something looks wrong, I can not found the semantic info for road line when using open3d script by follow the doc. Is that a bug or my mistake?

germanros1987, 2021-01-26:
@SimonLeeGit is this still an issue with 0.9.11?

fireofearth, 2021-05-23:
Hi @germanros1987, yes this is still a problem.
Many lane and street divider lines can be seen in semantic camera, but not semantic LIDAR.
Some street boundary lines and arrows on the road can be seen in semantic camera.

[image]

closed 2023-06-02 by stale[bot]