carla-simulator/carla #3639: I generated a pedestrian navigation mesh and the file path was correct, but the pedestrian could not be loaded

Yang-Zi97, 2020-11-27:
[image]

XGodina, 2020-12-01:
Hi@Yang-Zi97, 
Can you provide more information about it ? 
I suppose that you check our documentation right ?https://carla.readthedocs.io/en/latest/tuto_D_generate_pedestrian_navigation/#rebuild-the-navigation-binary

Yang-Zi97, 2020-12-02:
hi！xavi.
Thank you for your reply. 

We made the map using 3DMAX. According to step in the document, we have changed the names of all models, which are similar to Road_Road_Mesh_1...,
 using NoExport to export the unwanted model.
Using the executable program for Recast&amp;Detour,run:python spawn_npc.py.But only vehicles are generated, no pedestrians。After debugging, I know that pedestrians can generate pedestrians, but they will not appear on the navigation grid. I think there is something wrong with recast &amp; detour .

I hope you can help me.thinks

bernatx, 2020-12-02:
Hi, pedestrians can only spawn and walk on special areas (meshes) that have the proper nomenclature. So, you need to be sure that you have meshes starting with the name Road_Sidewalk, Road_Grass (optional) or Road_Crosswalk (optional). Pedestrians always spawn in areas named Road_Sidewalk, and they are the priorized ares where they walk. The Road_Grass are areas with less priority to walk but they will if the path is shorter walking on them, and the Road_Crosswalk are the areas they will walk if they need to cross a road. Be sure you have at least a Road_Sidewalk mesh. So:

- **Road_Sidewalk** is a must
- **Road_Grass** is optional
- **Road_Crosswalk** is optional

Regards,
Bernat

closed 2023-06-02 by stale[bot]