carla-simulator/carla #3643: Problem with vehicle dynamics for Audi E-Tron

alexk-92580, 2020-11-27:
Hi CARLA Team,

I have recently updated my installation of CARLA to version 0.9.10.1. With the update, I have noticed that the response of the Audi E-Tron to the throttle position has changed drastically. In particular, the maximal speed at full throttle has reduced to ~31.1m/s or 112 kph, from ~48.4 m/s before the update.

Another vehicle where I'm unsure whether it's working as intended (also before the update) is the Toyota Prius, which reaches a max. speed of ~18 m/s at 80% throttle, and does not accelerate further even at full throttle.  

Appreciate your help!
Alex

KunanonxGithub, 2020-11-29:
I'm interesting in this question.

jackbart94, 2021-11-18:
We updated all vehicles' physics in 0.9.13. Make sure to check them out!

closed 2021-11-18 by jackbart94