carla-simulator/carla #3644: Error in creating Carla project in Visual studio

YenTran32, 2020-11-28:
Hello, can someone help me get out of this problem?

I'm installing Carla on Window 7 64-bit. My process is: 

Step 1: installing additional software
Python 3.9 -> Cmake 3.19.1-> Git-2.29.2.2 -> Make 3.81 -> Visual Studio 2017 (including Windows 8.1 SDK and x64 Visual C++ Toolset) -> Unreal Engine 4.24.3

Step 2: running C:\Program Files\Git\git-cmd.exe and type in ‘git clone https://github.com/carla-simulator/carla’ then a folder is created in my computer ‘C:\Users\dellvn\carla’. After this, I renamed ‘carla’ to ‘Carla’. Then I download Assets from https://drive.google.com/open?id=10Tc0DMXpfH7W28zwagXsVTjpkhGx3Klz then I created the subfolder 'Unreal\CarlaUE4\Content\Carla' and extract the 'Content_0.9.4.tar.gz' to this 'Carla' folder

Step 3: setting environment variables
I set ‘Path’ variable to value ‘C:\MinGW\bin;C:\Users\dellvn\AppData\Roaming\Microsoft\Windows\Start Menu\Programs\Python 3.9;C:\Program Files (x86)\GnuWin32\bin;C:\CMake\bin’
and ‘Carla’ variable to value ‘C:\Users\dellvn\Carla’

Step 4: build a project in Visual studio
I go to ‘C:\Users\dellvn\Carla\Unreal\CarlaUE4’ and right click on ‘CarlaUE4.uproject’ -> select ‘Generate Visual Studio Project Files’. A file named ‘CarlaUE4.sln’ was generated. Then I open Visual studio 2017 -> File -> Open -> Project/Solution -> select the file ‘CarlaUE4.sln’. After that in Visual studio, I select ‘Build’ -> ‘Build solution’
The result is ‘Build failed’ and I got two errors:

Severity | Code | Description | Project | File | Line | Suppression State
-- | -- | -- | -- | -- | -- | --
Error |   | Unable to   instantiate module 'Carla': System.IO.DirectoryNotFoundException: Could not   find a part of the path 'C:\Users\dellvn\Carla\Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies\lib'. | CarlaUE4 | C:\Users\dellvn\Carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\UnrealBuildTool | 1 |  

Severity | Code | Description | Project | File | Line | Suppression State
-- | -- | -- | -- | -- | -- | --
Error | MSB3075 | The command   ""C:\Program Files\Epic Games\UE_4.24\Engine\Build\BatchFiles\Build.bat"   CarlaUE4Editor Win64 Development   -Project="C:\Users\dellvn\Carla\Unreal\CarlaUE4\CarlaUE4.uproject"   -WaitMutex -FromMsBuild" exited with code 5. Please verify that you have   sufficient rights to run this command. | CarlaUE4 | C:\Program Files   (x86)\Microsoft Visual   Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.MakeFile.Targets | 44 |  

[image]

Did I follow a correct procedure? How to solve this issue?

Thank you very much for any suggestion or comment.

BAROBA-3d, 2020-12-10:
Hi,
Did you do a build Pythonapi ?
Then you need to do Build launch
Then it should work.
I had the same errors, I fixed it with a redo of the Git-command.

closed 2021-03-05 by corkyw10

sushil-bhatta, 2022-11-04:
@YenTran32  Hi, how did you solved this issue ? I am having the same issue. I also don't need to use "make PythonAPI", because i want to use CARLA c++ client. 
Thanks in advance.