carla-simulator/carla #3645: Not able to compile successfully carla 0.9.10 on windows 10 make: *** [CarlaUE4Editor] ′í?ó 6

KNHP, 2020-11-28:
Not able to compile successfully carla 0.9.10 on windows 10
Error type:
make: *** [CarlaUE4Editor] ′í?ó 6

---------------------------------------------------------------------------------------------------------------------------------------

F:\carla>make launch
-[Setup]: [Batch params]: --boost-toolset msvc-14.1
-[Setup]: Asynchronous jobs: 8
-[Setup]: Boost toolset: msvc-14.1
-[Setup]: Install directory: "F:\carla\Build"
-[Setup]: Installing zlib...
-[install_zlib]: [Batch params]: --build-dir "F:\carla\Build"
-[install_zlib]: A zlib installation already exists.
-[install_zlib]: Delete "F:\carla\Build\zlib-install" if you want to force a rebuild.
-[install_zlib]: Exiting...
-[Setup]: Installing libpng...
-[install_libpng]: [Batch params]: --build-dir "F:\carla\Build" --zlib-install-dir "F:\carla\Build\zlib-install"
-[install_libpng]: A libpng installation already exists.
-[install_libpng]: Delete "F:\carla\Build\libpng-1.2.37-install" if you want to force a rebuild.
-[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
-[install_rpclib]: [Batch params]: --build-dir "F:\carla\Build"
-[install_rpclib]: A rpclib installation already exists.
-[install_rpclib]: Delete "F:\carla\Build\rpclib-install" if you want to force a rebuild.
-[install_rpclib]: Exiting...
-[Setup]: Installing Google Test...
-[install_gtest]: [Batch params]: --build-dir "F:\carla\Build"
-[install_gtest]: A Google Test installation already exists.
-[install_gtest]: Delete "F:\carla\Build\gtest-install" if you want to force a rebuild.
-[install_gtest]: Exiting...
-[Setup]: Installing "Recast & Detour"...
-[install_recast]: [Batch params]: --build-dir "F:\carla\Build"
-[install_recast]: A "Recast & Detour" installation already exists.
-[install_recast]: Delete "F:\carla\Build\recast-cdce4e-install" if you want to force a rebuild.
-[install_recast]: Exiting...
-[Setup]: Installing Boost...
-[install_boost]: [Batch params]: --build-dir "F:\carla\Build" --toolset msvc-14.1 --version 1.72.0 -j 8
-[install_boost]: A Boost installation already exists.
-[install_boost]: Delete "F:\carla\Build\boost-1.72.0-install" if you want to force a rebuild.
-[install_boost]: Exiting...
1 file(s) copied.
1 file(s) copied.
-[Setup]: Installing Xercesc...
-[install_xercesc]: [Batch params]: --build-dir "F:\carla\Build"
-[install_xercesc]: A xerces installation already exists.
-[install_xercesc]: Delete "F:\carla\Build\xerces-c-3.2.3-install" if you want to force a rebuild.
-[install_xercesc]: Exiting...
-[Setup]: Creating "CMakeLists.txt.in"...
-[Setup]:
###########
SUCCESS

###########

IMPORTANT!

All the CARLA library dependences should be installed now.
(You can remove all "*-src" folders in F:\carla\Build\ directory)

You only need the ASSET PACK with all the meshes and textures.

This script provides the assets for CARLA Latest:
You can download the assets from here:

   http://carla-assets.s3.amazonaws.com/20201103_4b3ab0c.tar.gz

Unzip it in the "F:\carla\Unreal\CarlaUE4\Content\Carla" folder.
If you want another version, search it in F:\carla\Util\ContentVersions.txt.

[... 59 lines cut ...]
C:\Program Files\Epic Games\UE_4.25\Engine\Source\Runtime\Engine\Classes\GameFramework/Actor.h(523): note: see declaration of 'AActor::Owner'
F:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/InertialMeasurementUnit.cpp(189): warning C4458: declaration of 'Stream' hides class member
F:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Sensor/Sensor.h(100): note: see declaration of 'ASensor::Stream'
F:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/ObstacleDetectionSensor.cpp(138): warning C4458: declaration of 'Episode' hides class member
F:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Sensor/Sensor.h(102): note: see declaration of 'ASensor::Episode'
F:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/PixelReader.cpp(96): error C2664: 'bool FPixelReader::WritePixelsToArray(UTextureRenderTarget2D &,TArray<FColor,FDefaultAllocator> &)': cannot convert argument 2 from 'TArray<PixelType,FDefaultAllocator64>' to 'TArray<FColor,FDefaultAllocator> &'
with
[
PixelType=FColor
]
F:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/SceneCaptureSensor.cpp(340): error C2039: 'AutoExposureCalibrationConstant': is not a member of 'FPostProcessSettings'
C:\Program Files\Epic Games\UE_4.25\Engine\Source\Runtime\Engine\Classes\Engine/Scene.h(613): note: see declaration of 'FPostProcessSettings'
F:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/SceneCaptureSensor.cpp(346): error C2039: 'AutoExposureCalibrationConstant': is not a member of 'FPostProcessSettings'
C:\Program Files\Epic Games\UE_4.25\Engine\Source\Runtime\Engine\Classes\Engine/Scene.h(613): note: see declaration of 'FPostProcessSettings'
F:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/SceneCaptureSensor.cpp(544): error C2039: 'bOverride_AutoExposureCalibrationConstant': is not a member of 'FPostProcessSettings'
C:\Program Files\Epic Games\UE_4.25\Engine\Source\Runtime\Engine\Classes\Engine/Scene.h(613): note: see declaration of 'FPostProcessSettings'
F:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Walker/WalkerController.cpp(28): warning C4458: declaration of 'Character' hides class member
C:\Program Files\Epic Games\UE_4.25\Engine\Source\Runtime\Engine\Classes\GameFramework/Controller.h(76): note: see declaration of 'AController::Character'
F:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Walker/WalkerController.cpp(83): warning C4458: declaration of 'Character' hides class member
C:\Program Files\Epic Games\UE_4.25\Engine\Source\Runtime\Engine\Classes\GameFramework/Controller.h(76): note: see declaration of 'AController::Character'
F:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Walker/WalkerController.cpp(112): warning C4458: declaration of 'Character' hides class member
C:\Program Files\Epic Games\UE_4.25\Engine\Source\Runtime\Engine\Classes\GameFramework/Controller.h(76): note: see declaration of 'AController::Character'
F:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Walker/WalkerController.cpp(126): warning C4458: declaration of 'Character' hides class member
C:\Program Files\Epic Games\UE_4.25\Engine\Source\Runtime\Engine\Classes\GameFramework/Controller.h(76): note: see declaration of 'AController::Character'
make: *** [CarlaUE4Editor] ′í?ó 6

BAROBA-3d, 2020-12-06:
Hi,
You should use 4.24 instead of 4.25 and VS 2017.

corkyw10, 2021-03-05:
Duplicate of #3640

closed 2021-03-05 by corkyw10