carla-simulator/carla #3649: Some general questions

AvivSham, 2020-11-30:
Realy awesome repo! shout out to all maintainers!
I have several questions related to Carla:

- Is it possible to create a deterministic scene? i.e. create a scene which will repeat itself in terms of movement directions of cars but with the ability to change car types, surrounding, etc.
- Is it possible to render videos of autonomous driving using Carla? or it functions as a computer game?
- If the latter is feasible can it be done in different resolutions and other parameters such as FPS?
- How long will it take to render and save 1 minute duration video on 3080ti (approximately)

Thank you in advance for helping.

ebadi, 2020-12-03:
Replaying scenarios are possible. Search for "replay" in the documentations.

AvivSham, 2020-12-03:
I expected the answer to the first question to be "yes," but what about the others?
@ebadi  are you one of the maintainers?

ebadi, 2020-12-04:
No, I am not a maintainer.

PTorr, 2021-03-04:
You can do all of the above.
The recording time depends on resolution/ num of actors etc.

AvivSham, 2021-03-18:
@DSantosO | @PyTorr | @ebadi  Thank you for your response.
Another question - I want to integrate Carla with project acoustics to do so I need to work with Carla in UE. Is there any straight forward to use Carla in UE? (Addon or something like that?)
@dvirsamuel

DSantosO, 2021-03-30:
Hello @AvivSham,
Yes, just download from the source and you can work directly in the Unreal Engine editor. Check in our [documentation website](https://carla.readthedocs.io) in the Linux/Windows build sections.

closed 2021-03-30 by DSantosO

AvivSham, 2021-04-01:
@DSantosO Thank you for the response.
If I understand you correctly I should follow [these instructions ](https://carla.readthedocs.io/en/0.9.11/build_linux/#unreal-engine) isn't it?
@dvirsamuel