carla-simulator/carla #3650: Set traffic manager to synchronous mode cause skip frame issue

eherozhao, 2020-12-01:
Hi, I am using Synchronous mode to run this simulation. The code is written depending on [example](https://github.com/carla-simulator/carla/blob/6d20f7d3742460ae300d19a3543518416b60925b/PythonAPI/examples/synchronous_mode.py). However, when I set a hero player and auto_pilot it, the frame is not continuous and skipping frame issue happens. When commenting `traffic_manager.set_synchronous_mode(True)`, the frame changed back to be continuous. I notice in the [test_sync.py](https://github.com/carla-simulator/carla/blob/6d20f7d3742460ae300d19a3543518416b60925b/PythonAPI/test/smoke/test_sync.py#L87), it comment the `set_autopilot` line. I am not sure whether it is an issue inside the traffic_manager. Please take a look at it. Thank you.

corkyw10, 2021-03-08:
@jackbart94 could you have a look at this please?

jackbart94, 2021-03-10:
Could you try with 0.9.11 or latest CARLA that this is indeed still happening? Also, the synchronous_mode.py script is relatively outdated, could you try and see if it happens in synchronous mode with spawn_npc.py for example?

eherozhao, 2021-05-16:
My use case was similar as running `manual_control.py` in the sync mode so that I could just run given simulation time. I didn't find the sync mode support for this file.

jackbart94, 2021-11-18:
> My use case was similar as running `manual_control.py` in the sync mode so that I could just run given simulation time. I didn't find the sync mode support for this file.

Now it's available by doing ``--sync``.

closed 2021-11-18 by jackbart94