carla-simulator/carla #3651: Lane invasion detector unable to detect lane markings in time

tinmodeHuang, 2020-12-01:
I can conclude from the official documentation that the space between the four wheels is considered as a whole to detect whether the vehicle invade other lane or not. even so, I haven't ever known about the reason why Lane invasion detector unable to detect curb all the time and even white road sideline sometimes.

[image]
my implementation as follows:
```python
LaneInvasionSensor = self.world.spawn_actor(sensor, carla.Transform(), self.ego_vehicle)
self.sensors_list.append(LaneInvasionSensor)
LaneInvasionSensor.listen(lambda event: self.on_invasion(event))
...
if [lane_marking for lane_marking in self.lane_markings if lane_marking in ['Solid','SolidSolid','Curb','Grass']]:
    done = True    # it means that an episode comes to end during training

def on_invasion(self, event):
    lane_types = set(x.type for x in event.crossed_lane_markings)
        for x in lane_types:
            self.lane_markings.append(str(x))
```

tinmodeHuang, 2021-01-30:
@jackbart94 if convient, it is the best that you could help me implement it, which also gives users more material to reference, but that seems impossible, if that way, I wonder how to acquire the edge of road

jackbart94, 2021-03-10:
Well, the LaneInvasionSensor is a bit broken, so this is what you can do:

- use the get_waypoint() in the location of the vehicle, setting the project_to_road bool to False. This will make sure that you get it at the center of the vehicle.
- then, you want to again use get_waypoint() to find the closest lane marking, and then compute the distance between the two.

You might want to check the [Python API information and code snippet in the documentation](https://carla.readthedocs.io/en/latest/python_api/#getters_9) to do this. There's all the information you need.

GyChou, 2021-04-04:
Hi,

I found that when the ego vehicle enters into a junction, the yaw of the current waypoint does not match the direction of the target lane, for example, I plan the route to turn left, but the yaw of the waypoint points to straight ahead.

The way I get the current waypoint is:
world.get_map().get_waypoint(vehicle.get_location())

I doubt that this method: world.get_map().get_waypoint(vehicle.get_location()) is to get the nearest waypoint of vehicle.get_location(), due to the waypoints of each direction of the junction are very close, resulting in the selection of the wrong lane waypoint.

I holp get the waypoint of my current location in a junction and its yaw matches my target lane, how do I do that?

I used Carla0.9.11.

Thanks for your help

jackbart94, 2021-04-06:
You should be able to get your current waypoint with whatever structure you are using to compute the path of your vehicle. If you're using the Traffic Manager, you'll have to look into the Localization Stage.

GyChou, 2021-04-09:
Yes，I could get my current waypoint by ' world.get_map().get_waypoint(vehicle.get_location())', but the current waypoint **yaw** is wrong in the junction sometimes.

closed 2021-11-18 by jackbart94