carla-simulator/carla #3652: vehicle.is_at_traffic_light

NurbekHalikulov, 2020-12-01:


NurbekHalikulov, 2020-12-01:
I am creating a basic code in carla editor as I am a newbie here. What I want to do is to spawn a car in some pre-defined location in town 05 and check whether the traffic light is getting green when it is red. I am going to try to simulate ''Traffic Light Recipe'' from https://carla.readthedocs.io/en/latest/ref_code_recipes/

Here, I found some materials in carla world and I am going to provide them here. At the end, I am not getting any result, unfortunately.

So, here we go. I am choosing a pre-defined location for the tesla vehicle, i.e. carla.Location(x=8, y=6, z=3), which is staying close to a traffic light. I tried to give autopilot_mode to the tesla. I can also apply a control to it. Which one I should use here to test my recipe code?

I tried with two types of source codes to run the program. But both of them is not working, i.e. my traffic light is not turning into green when it is red. 

COULD YOU PLEASE HELP ME TO FIX THIS ISSUE?

    try:
        client = carla.Client(args.host, args.port)
        client.set_timeout(2.0)

        world = client.get_world()

        vehicle_bp = random.choice(world.get_blueprint_library().filter('vehicle.tesla.model3'))
        vehicle_bp.set_attribute('color', '255,0,0')

        #transform = random.choice(world.get_map().get_spawn_points())

        transform = carla.Transform(carla.Location(x=12, y=6, z=3), carla.Rotation())

        vehicle = world.try_spawn_actor(vehicle_bp, transform)

        vehicle.apply_control(carla.VehicleControl(throttle=5.0, steer=0.0))
        #vehicle.set_autopilot()
        actor_list.append(vehicle)
       
        ###### FIRST TRY WITH THIS SOURCE CODE, BUT IT IS NOT WORKING ######
        if vehicle.is_at_traffic_light():
            tl = vehicle.get_traffic_light()
            for x in tl.get_group_traffic_lights():
                x.freeze()
                if x.get_pole_index() == tl.id:
                    state = carla.TrafficLightState.Green
                else:
                    state = carla.TrafficLightState.Red
                x.set_state(state)
          

          ###### SECOND TRY WITH THIS SOURCE CODE, BUT IT IS NOT WORKING ######
        traffic_lights = world.get_actors().filter('traffic.traffic_light')

        for i in traffic_lights:
            if vehicle.is_at_traffic_light():
                traffic_light = vehicle.get_traffic_light()
                if traffic_light.get_state() == carla.TrafficLightState.Red:
                    traffic_light.set_state(carla.TrafficLightState.Green

        time.sleep(args.tick_time)

    finally:

        print('destroying actors')
        for actor in actor_list:
            actor.destroy()
        print('done.')

moein1924, 2020-12-02:
You should put the code inside the loop. for example in manual_control.py you cab try this:

        while True:
            clock.tick_busy_loop(60)
            if controller.parse_events(client, world, clock):
                return

            if world.player.is_at_traffic_light():
                traffic_light = world.player.get_traffic_light()
                print(traffic_light)
                if traffic_light is not None and traffic_light.get_state() == carla.TrafficLightState.Red:
                    # world.hud.notification("Traffic light changed! Good to go!")
                    traffic_light.set_state(carla.TrafficLightState.Green)

moein1924, 2020-12-02:
You should put the code inside the loop. for example in manual_control.py you can try this:

        while True:
            clock.tick_busy_loop(60)
            if controller.parse_events(client, world, clock):
                return

            if world.player.is_at_traffic_light():
                traffic_light = world.player.get_traffic_light()
                print(traffic_light)
                if traffic_light.get_state() == carla.TrafficLightState.Red:
                    # world.hud.notification("Traffic light changed! Good to go!")
                    traffic_light.set_state(carla.TrafficLightState.Green)

closed 2020-12-03 by NurbekHalikulov