carla-simulator/carla #3653: [Camera RGB][synchronous mode]:

Vaan5, 2020-12-01:
### Problem - short description

I have observed weird (atleast to me) behavior when using `sensor_tick` of the camera sensor (didn't try others yet) and the synchronous mode. 

* As soon as a value is set (different than 0), the sensor data is not being transmitted during the second tick.
* Additionally, the sensor data doesn't get transmitted according to the least common multiple

See Observations for more info.

### Example script
This is a modified `sensor_synchronization.py` example:
```
import glob
import os
import sys
from queue import Queue
from queue import Empty

sys.path.insert(0, r'C:\workdir\installations\carla\0.9.9.4\CARLA_0.9.9.4\WindowsNoEditor\PythonAPI\carla\dist\carla-0.9.9-py3.7-win-amd64.egg')

import carla

def sensor_callback(sensor_data, sensor_queue, sensor_name):
    # just to make sure that something weird doesn't happen due to the queue
    print("RECEIVED sensor data for frame: {}".format(sensor_data.frame))

    sensor_queue.put((sensor_data.frame, sensor_name, sensor_data.timestamp))

def main():
    # We start creating the client
    client = carla.Client('localhost', 2000)
    client.set_timeout(2.0)
    world = client.get_world()

    try:
        original_settings = world.get_settings()
        settings = world.get_settings()

        # We set CARLA syncronous mode
        settings.fixed_delta_seconds = 0.02
        settings.synchronous_mode = True
        world.apply_settings(settings)

        sensor_queue = Queue()

        blueprint_library = world.get_blueprint_library()
        cam_bp = blueprint_library.find('sensor.camera.rgb')

        sensor_list = []

        ## 1. no sensor tick changes
        #cam_bp.set_attribute('sensor_tick', '0.0')  # 2. manually set to 0.0
        #cam_bp.set_attribute('sensor_tick', '0.02')  # 3. manually set to 0.02 (same as fixed_delta_seconds)
        #cam_bp.set_attribute('sensor_tick', '0.04')  # 4. manually set to 0.04 (double the value of fixed_delta_seconds)
        #cam_bp.set_attribute('sensor_tick', '0.03')  # 5. manually set to 0.03 (least common multiple is 0.06)
        cam01 = world.spawn_actor(cam_bp, carla.Transform())
        cam01.listen(lambda data: sensor_callback(data, sensor_queue, "camera01"))
        sensor_list.append(cam01)

        # Main loop
        for i in range(10):
            # Tick the server
            world.tick()
            w_frame = world.get_snapshot().frame
            w_time = world.get_snapshot().timestamp
            print("\nWorld's frame: %d timestamp: %f" % (w_frame, w_time.elapsed_seconds))

            try:
                for i in range(0, len(sensor_list)):
                    s_frame = sensor_queue.get(True, 1.0)
[... 240 lines cut ...]
6.  0.10 -> 0.12    MISS
7.  0.12 -> 0.14    HIT
8.  0.14 -> 0.16    MISS
```
3. In case 5, why do we have a sequence of HIT MISS HIT MISS at the beginning? I would expect something like:
```
1.  0.00 -> 0.02    HIT (if I interpret this as an initial value)
2.  0.02 -> 0.04    MISS (0.03 seconds didn't pass from 0.02 -> 0.04)
3.  0.04 -> 0.06    HIT (at 0.05, 0.03 seconds passed)
4.  0.06 -> 0.08    HIT (at 0.08 0.03 seconds passed again)
5.  0.08 -> 0.10    MISS
6.  0.10 -> 0.12    HIT
7.  0.12 -> 0.14    HIT
8.  0.14 -> 0.16    MISS
```

Could it be related to how the ticks are stored (double vs float) ?

### Related issues
* #3385 and the related PR and testing issue

### Environment
**Platform**: Windows 10
**Python**: Python 3.7.7
**Carla**: 0.9.7.2, 0.9.9.4, 0.9.10 (ignored due to the linked issue)

OmarAbdElNaser, 2020-12-15:
Is there a solution for this issue without waiting till the next release ?

HaoZhouGT, 2021-02-04:
Hi, I found a similar issue in Carla 0.9.10, changing the sensor tick for the camera just doesn't work. The timestamps of frames do not change accordingly after we set a different sensor tick.  

Do you find any solution?

OmarAbdElNaser, 2021-02-05:
at the latest version 0.9.11 they solved this issue, but i didn't try it yet.

HaoZhouGT, 2021-02-05:
> at the latest version 0.9.11 they solved this issue, but i didn't try it yet.

Thanks. I also tried the async mode, it does not work either. And the real sensor tick depends on the quality of rendering. Did you encounter a similar issue?

OmarAbdElNaser, 2021-02-05:
sorry i didn't go through this issue before, but i checked the bugs they solved in the latest version, and they mentioned it.

HaoZhouGT, 2021-02-05:
@OmarAbdElNaser Thanks. Where did they mention it's fixed? Can you direct me? I saw something like `Fixed bug causing camera-based sensors to stop sending data`, not sure it is related though.

OmarAbdElNaser, 2021-02-05:
This is what i'm refering to, the camera issue was sending data without caring about the sensor tick, they said they solved it.

HaoZhouGT, 2021-02-05:
@OmarAbdElNaser Great, I will test it with the new version and let you know if it's really fixed. Thanks.

HaoZhouGT, 2021-02-07:
@OmarAbdElNaser I tested 0.9.11, the sensor_tick functionality has been improved, at least in the synchronized mode. I see the sensor tick setting is effective if you don't choose a very small value, this is expected due to the performance limitation of our system I guess.

closed 2021-03-08 by corkyw10

Vaan5, 2022-10-12:
This is not fixed. Problem still persists.

e.g. in case 4 from my description, we still get 2 misses one after the other.

Tried with 0.9.13.

Please reopen the issue.