carla-simulator/carla #3654: PyGame scene runs very slow fps

SeymurSm, 2020-12-01:
Hello, everyone,

I am trying to run Carla at Ubuntu 18.04  with Nvidia 2070 Graphics card.   Currently, I am at the beginning and I have tried to run examples such as manual_control.py. Everything works but the scene runs on PyGame is extremely slow (~1fps). I think there should be some option to handle this issue on the Unreal side or some options at the start to run the simulator. Does anyone face the same problem before?   

Thanks in advance

NurbekHalikulov, 2020-12-02:
Hi, 

If the simulation is running at very low FPS rates, go to Edit/Editor preferences/Performance in the UE editor and disable or uncheck **Use less CPU when in background**. 
Source: https://carla.readthedocs.io/en/latest/build_windows/ 

It helped me, maybe it will help you also

Cheers

charlie-wartnaby, 2020-12-02:
This doesn't help you, but I find I get a good frame rate in pygame on an Ubuntu 18.04 machine also with an Nvidia RTX 2070 GPU.
However, the same version in Windows (also with an Nvidia GPU, maybe 2060??) has a good frame rate in Unreal but a very poor rate, like yours, in Pygame. So I seem to get hardware acceleration (I guess) of pygame under Ubuntu but not under Windows.

goksanisil23, 2020-12-05:
Hi, I also have low frame rates in the Pygame window on Ubuntu 18 with Nvidia 2080Ti.
One interesting thing that I observed is that, when pygame is rendering the camera, if you click on some other window than Pygame (like Unreal window instead), Pygame FPS goes back to 60. But as soon as you interact with the pygame to, lets say manually control the vehicle, it goes back to 3 FPS.

Would be good to know if others observe this phenomenon as well.

closed 2023-06-02 by stale[bot]