carla-simulator/carla #3655: odometer for the agent: computing accumulated distance travelled by the spawned vehicle in the world

Horcrux1, 2020-12-01:
I wanted to look for an API method that computes the distance that the agent(vehicle) traveled int the world in km. Exactly as what an odometer in vehicles does. There is no built function that I could find in the documentation, can you guys help me with this ? Also, I know that I can write a code for it myself, however, I am still interested in having an easy method provided by Carla doing this for me. 

I very much appreciate your answers.

KangChou, 2020-12-23:
I have a calculation method here（~~~go to carla-hello）

corkyw10, 2021-03-08:
@Axel1092 could you have a look at this please?

Axel1092, 2021-03-15:
Hi @Horcrux1, we don't currently have any function to record the distance traveled by a vehicle. You will need to add up the distance traversed by the agent each frame yourself from the client side.